RT-Thread/rt-thread #6143: cmake 工程include path 重排序问题【4.11】

tonyzhangnxp, 2022-07-06:
hello there

cmake工程，include path会按字母重新排序，这会引起不同目录下相同头文件引用错误。
经过分析，tools/utils.py 脚本里再处理CPPPATH 去除相同项操作导致了list乱序，后边又调用了sort排序。
原代码：
`
 # remove repeat path
        paths = [i for i in set(CPPPATH)]
        CPPPATH = []
        for path in paths:
            
            if PrefixPath(RTT_ROOT, path):
                CPPPATH += [os.path.abspath(path).replace('\\', '/')]

            elif PrefixPath(BSP_ROOT, path):
                CPPPATH += [os.path.abspath(path).replace('\\', '/')]

            else:
                CPPPATH += ['"%s",' % path.replace('\\', '/')]
        
        CPPPATH.sort()
`

应该改为：
 # remove repeat path
        paths = []
        for p in CPPPATH:
            if p not in paths:
                paths.append(p)

        CPPPATH = []
        for path in paths:
            
            if PrefixPath(RTT_ROOT, path):
                CPPPATH += [os.path.abspath(path).replace('\\', '/')]

            elif PrefixPath(BSP_ROOT, path):
                CPPPATH += [os.path.abspath(path).replace('\\', '/')]

            else:
                CPPPATH += ['"%s",' % path.replace('\\', '/')]

BR

mysterywolf, 2022-07-06:
@liukangcc 康康麻烦看看这个问题

liukangcc, 2022-07-06:
all changes have been test

mysterywolf, 2022-08-02:
感谢 已经合并

closed 2022-08-02 by mysterywolf