RT-Thread/rt-thread #6183: [CPK-RA6M4] 用户自定义看门狗溢出时间无效

kurisaW, 2022-07-22:
CPK-RA6M4评估板结合FSP及RT-Thread Studio工具，可方便开发者对嵌入式系统应用程序进程开发。

在学习RT-Thread设备驱动时，注意到RT-Thread提供`rt_err_t rt_device_control(rt_device_t dev, rt_uint8_t cmd, void* arg);`API接口对看门狗溢出时间进行配置，所以在测试时设置了看门狗的溢出时间，当初始化配置FSP后，进行编译，却发现程序并没有按照所编写的代码在溢出时间后喂狗。

然而我在这里发现了问题：
点击打开drv_wdt.c（路径：/libraries/HAL_Drivers/drv_wdt.c），可以看到在设置看门狗超时时间时使用了break命令，因此我们用户设定的超时时间被过滤。

总结：在CPK-RA6M4中看门狗超时时间的设定是在FSP中完成初始化的，当然如果想在RT-Thread中设置，可选择修改drv_wdt.c中关于看门狗超时时间的相关代码。

[image]

mysterywolf, 2022-07-30:
感谢反馈  这里没有正确写入错误码

ShermanShao, 2022-08-03:
感谢反馈，在瑞萨的RA库中，WDT部分如果需要修改timeout得在FSP配置工具中操作。API层面没有对这部分修改的支持。所以这里可能得返回一个错误码。
关于这部分你有什么好的建议吗？@kurisaW

mysterywolf, 2022-08-07:
我感觉可以通过LOG_W进行警示并直接返回错误码即可

mysterywolf, 2022-08-15:
https://github.com/RT-Thread/rt-thread/pull/6279

closed 2022-08-15 by mysterywolf