RT-Thread/rt-thread #6186: The board is stuck after powering down and then powering on, and the board freezes in the spi flash initialization after pressing reset.

Joiey, 2022-07-23:
The board is stuck after powering down and then powering on, and the board freezes in the spi flash initialization after pressing reset. Re-burning other normal Apps after resetting and then burning the App function can work normally.
板子掉电再上电后卡死，按复位后卡住在spi flash初始化。**重新烧录其他正常App复位后再烧录该App功能都可正常运行。**

RTThread版本4.0.5，MCU使用F411CC，onchip flash 256KB，RAM 128KB。

```
 \ | /
- RT -  Thread Operating System 
/ | \     4.0.5 build 2022 
2006 - 2021 Copyright by rt-thread team
do components initialization.initialize rti_board_end:0 done
initialize ulog_console_backend_init:0 done
initialize rt_hw_rtc_init:0 done
initialize stm32_pwm_init:0 done
initialize rt_i2c_core_init:0 done
initialize rt_hw_spi_flash_init
```

Joiey, 2022-07-23:
逐行调试问题解决了。
问题是由于在 `ulog` 开启了 `syslog`，导致函数 `sfud_device_init()` 调用宏 `SFUD_INFO()`输出卡死。关闭 `syslog` 即可解决。
从源码看 `ulog` 在板级 `INIT_BOARD_EXPORT(ulog_init)` 就已经初始化，但是在 `syslog` 中`openlog()` 不知道何时初始化，虽然文档中描述`openlog()`是可选的并会在第一次调用时自动打开。
希望能解决该 `syslog` bug。

Guozhanxin, 2023-03-24:
> 

有什么修复的建议吗？