RT-Thread/rt-thread #6189: 串口不会通过导出自动初始化

helloobaby, 2022-07-24:
uart_init需要手动调用。

环境是vscode+gcc 板子是stm32f407 

对应官方文档https://www.rt-thread.org/document/site/#/rt-thread-version/rt-thread-nano/finsh-port/an0045-finsh-port
我也是nano的。

https://www.rt-thread.org/document/site/#/rt-thread-version/rt-thread-nano/finsh-port/an0045-finsh-port
[image]

mysterywolf, 2022-07-24:
rt_hw_board_init 函数调用了吗？

helloobaby, 2022-07-25:
调用了。

void rt_hw_board_init(void)
{
    extern void SystemClock_Config(void);
    
    HAL_Init();
    SystemClock_Config();
    SystemCoreClockUpdate();
    uart_init(); //我加的,这样串口就初始化了
}

解决这个问题并不麻烦，就是知道这个问题比较费劲。

nongxiaoming, 2022-07-26:
在rt_hw_board_init 函数中需要加入这么一行才能初始化INIT_BOARD_EXPORT导出的函数。
```c
    /* Board underlying hardware initialization */
#ifdef RT_USING_COMPONENTS_INIT
    rt_components_board_init();
#endif
```

closed 2022-07-26 by mysterywolf