RT-Thread/rt-thread #6190: RT1170-EVK BSP 编译错误

xjy198903, 2022-07-25:
RT1170-EVK在4.1.1版本中，编译失败，无论是keil或是IAR，都会找不到components\libc\compilers\common下的stdlib.c和ctime.c；增加CAN支持后，drv_can.c编译失败

xjy198903, 2022-07-25:
另外，RT1170-EVK在rt-thread是否能实现多核通信，比如使用eRPC来进行RPMSG

mysterywolf, 2022-07-25:
你重新生成工程了嘛？

xjy198903, 2022-07-26:
> 你重新生成工程了嘛？
重新生成工程了，compilers那个问题没有了，不过CAN工程还是编译不过，需要手动修改源文件及SConscript

thewon86, 2022-07-27:
> 

您好，我也有打算用 erpc ，做点儿东西。但是没找到在rtos上使用的教程

xjy198903, 2022-07-27:
增加ETH支持时，drv_eth.c报错：
[image]

xjy198903, 2022-07-27:
> > 
> 
> 您好，我也有打算用 erpc ，做点儿东西。但是没找到在rtos上使用的教程

我在看能不能借用nxp官方的erpc的Demo

mysterywolf, 2022-07-27:
目前1170还没有支持Can https://github.com/RT-Thread/rt-thread/blob/master/bsp/imxrt/imxrt1170-nxp-evk/README_zh.md

xjy198903, 2022-07-27:
> 目前1170还没有支持Can https://github.com/RT-Thread/rt-thread/blob/master/bsp/imxrt/imxrt1170-nxp-evk/README_zh.md

文档里写的支持Ethernet，但实际编译都有问题：
[image]

andychen329, 2022-08-01:
@xinyigao @xinyigao11 看看哦

xjy198903, 2022-08-01:
> @xinyigao @xinyigao11 看看哦
网络设备引脚也没配对，这个版本不应该称为release版本

xinyigao, 2022-08-01:
好的

Caution: EXT Email

xjy198903, 2022-08-08:
问下，RT1170的正式版大概什么时候能出？目前千兆网络不通，SDRAM初始化成功，但使用rt_malloc时，申请大于1M的空间，串口会卡住不动，无法再进行操作；单独使用rt_memheap_alloc可以申请1M的空间

xinyigao, 2022-08-08:
1170目前支持100M ENET，enet_rst引脚为（GPIO6 ，12）在设置的时候需要注意，代码最近几天就会更新，至于您说的rt_malloc申请问题，需要具体分析，同时还要看下代码中分配的堆大小

Caution: EXT Email

xinyigao, 2022-08-08:
设备的引脚是在menuconfig中配置的，同时还要使能gpio模块才能正确初始化

xjy198903, 2022-08-08:
百兆的我编过了，禁用DCHP，使用默认IP 192.168.1.30，但是ping不通开发主机；我是在main函数中进行的rt_malloc的问题，发现不止是不能分配一兆。分布131072大小就会卡住，系统无法再进行操作

xinyigao, 2022-08-08:
***@***.***

Caution: EXT Email

xjy198903, 2022-08-08:
这是啥？

xinyigao, 2022-08-08:
这边经过测试
“
#include <rtthread.h>
#include <rtdevice.h>
#include "drv_gpio.h"
char b_path[141072]={1};
int main(void)
{
#ifndef PHY_USING_KSZ8081
    while (1)
    {
         char *full_path;
          full_path = (char *)rt_malloc(141072);
rt_memcpy(full_path, b_path, 141072);
          rt_kprintf("test\r\n");
           rt_thread_mdelay(500);
    }
#endif
}
“
在main函数中添加rt_malloc，可以正常运行并打印信息

Caution: EXT Email

这是啥？

xjy198903, 2022-08-08:
加了SDRAM的支持吗？

closed 2022-08-08 by mysterywolf

mysterywolf, 2022-08-08:
@xinyigao 感谢帮忙测试~

xinyigao, 2022-08-09:
如果您是想使用0x80 开头的地址，在keil中已经实现可以直接使用（不用配置sdram）,如果想在iar中直接使用，可以修改link.icf文件（直接参考官方sdk中带sdram配置的.icf 文件）

Caution: EXT Email

xjy198903, 2022-08-09:
> 

麻烦确认下目前已发布的1170bsp百兆网络驱动部分，在不修改源码的情况下，编译都无法通过，我在手动修改编译错误的地方后，镜像可以生成，但网络不通，测试方法是ping主机，双向互ping均不通

mysterywolf, 2022-08-09:
https://github.com/RT-Thread/rt-thread/pull/6244 这有一个新的PR 等合并之后你再试一下

xjy198903, 2022-08-10:
> #6244 这有一个新的PR 等合并之后你再试一下

已参考并修改，编译测试一把过，感谢！现在要琢磨千兆网络了，这个才是我真正需要的

xjy198903, 2022-08-10:
NXP官方有支持千兆phy(rtl8211f)的计划吗？

xinyigao, 2022-10-11:
这边经过测试添加sdram后也是支持的
您要不下是内存越界还是其他原因造成的？

xjy198903, 2022-10-11:
这个问题已经处理完了