RT-Thread/rt-thread #6234: How to create a new i2c project to stm32f469 discovery kit to see Terminal (Logs)

neuberfran, 2022-08-06:
Hi,

I have SmartDrive in C++ (to PlatFormIo) AND in Java to Android

https://github.com/neuberfran/SmartDriveEmC
https://github.com/neuberfran/SmartDrive

I need help to transpose this driver to rt-thread/stm32f469 disco kit

I saw it on this site: https://club.rt-thread.org/index.php/ask/question/0d6fd0d49aa2f1af.html
that the I2C1 (Bus) is already used by the screen that comes with the kit, **correct**?

I tried several sites, like the one below:
https://www.rt-thread.io/document/site/rtthread-studio/drivers/soft-i2c/rtthread-studio-soft-i2c/

But I just don't get the **msh>** terminal when I create project in Base On MCU. Only in Base On Board. Why? 

How to solve it?

mysterywolf, 2022-08-06:
We have fixed this problem. STM32 F469 DISC hardware has two different types with two different touch sensor types. Please follow this video to import the latest code (note latest release version) to RT-Thread Studio. https://www.youtube.com/watch?v=fREPLuh-h8k

mysterywolf, 2022-08-06:
latest code (not latest release version)

neuberfran, 2022-08-06:
@mysterywolf Not solved yet. Look pic. The new project named discovery6 not have the drivers folder. I need it to configure i2c2 in board.h file. 

This project (discovery6) looks like this in the same way as before when generating a project using the Base on BOARD option. Being that to appear the drivers folder it is necessary to generate the project by Base on MCU
[image]

mysterywolf, 2022-08-06:
[image]
[image]
[image]
[image]

neuberfran, 2022-08-06:
@mysterywolf How can I will use I2C2 like this:
https://chowdera.com/2022/159/202206080317229229.html

?

mysterywolf, 2022-08-06:
Then, select the "hammer" to compile the project as well as auto-save the configurations.

mysterywolf, 2022-08-06:
Not necessary. This buttons can auto-select all of the dependency, like you mentioned, iic and others. By using these buttons, users can one-step configure the board.

neuberfran, 2022-08-06:
@mysterywolf I have issue here to add lvgl:

../applications/lvgl/lv_port_indev.c:22:20: fatal error: ft6236.h: No such file or directory
compilation terminated.
make: *** [applications/lvgl/subdir.mk:21: applications/lvgl/lv_port_indev.o] Error 1

[image]

neuberfran, 2022-08-06:
@mysterywolf 
[image]

mysterywolf, 2022-08-06:
Ohhhmy problem there is a index error in my side. Give me some time

mysterywolf, 2022-08-06:
Hi please close the RT-Thread Studio and reopen it. Wait for 10 secs, and then right click
[image]
and then compile the project again.

neuberfran, 2022-08-07:
@mysterywolf Now I want (step to step) to enter smartdrive in msn/> and I will can use My external i2c driver. I talked about it at the beginning of this topic. `[]s

https://github.com/neuberfran/SmartDriveEmC
https://github.com/neuberfran/SmartDrive

mysterywolf, 2022-08-07:
If you want to export a command, you can follow this branch of code: 
https://github.com/RT-Thread/rt-thread/blob/master/components/finsh/cmd.c#L891-L992
use `MSH_CMD_EXPORT_ALIAS` to export

neuberfran, 2022-08-07:
@mysterywolf Sorry to insist. I somehow consider myself a beginner. I need to do it step by step:

1) Why did you send me this link? What will I proceed from it?

[image]
[image]

mysterywolf, 2022-08-07:
So you want to add another I2C ?

neuberfran, 2022-08-07:
@mysterywolf Yes. I have external driver. Photos above.

These links below show that the driver works on esp32/platformio and Java/Android. But I need to make modifications and implementations. Which?

https://github.com/neuberfran/SmartDriveEmC
https://github.com/neuberfran/SmartDrive****

mysterywolf, 2022-08-07:
https://github.com/neuberfran/SmartDriveEmC This is Arduino library right?

RT-Thread support Arduino API. There is a compatibility layer of Arduino for RT-Thread is called RTduino. and STM32F469 also Support it.
[image]

More information please see: https://github.com/RTduino/RTduino#3-arduino%E5%BA%93%E7%9A%84%E5%AF%BC%E5%85%A5%E5%92%8C%E4%BD%BF%E7%94%A8

mysterywolf, 2022-08-07:
You can directly copy this library to `user` this folder and right click `sync sconscript to project`
[image]

mysterywolf, 2022-08-07:
[image]
There is the setup and loop function

mysterywolf, 2022-08-07:
https://github.com/RT-Thread/rt-thread/tree/master/bsp/stm32/stm32f469-st-disco/applications/arduino_pinout

neuberfran, 2022-08-07:
@mysterywolf Tks. I stay with issue in this tutorial too(for 2 weeks now):
https://blog.csdn.net/bonson2004/article/details/122040352

In ifconfig part I can see (using ifconfg e0 Not ifconfig Only egual tutorial) 192.168.1.30, etc, etc...But I can't ping to 192.168.1.2. I don't know why. Do u know

[image]

mysterywolf, 2022-08-07:
You cannot direct connect the network cable with computer and board, you need to use a router.

neuberfran, 2022-08-07:
@mysterywolf But from what I understand from the tutorial the guy is connecting the ch32v307 directly to his notebook. Not even about DHCP he talked about in the tutorial.

mysterywolf, 2022-08-07:
He missed. You need to use cross over network cable rather than a normal (straight-through) network cable if you want to directly connect the board with computer; otherwise, you need to use a router.

neuberfran, 2022-08-07:
@mysterywolf There is something wrong with the programming of the csds project. In this photo below, I connected a red cable to the PC and turned on the switch light. On the other hand, the yellow cable connected to the ch32v307 does not light up anything on the switch. will never **ping.**

[image]

neuberfran, 2022-08-07:
> [image] There is the setup and loop function

Will I be able to use i2c2 with this system (arduino pinout)? (Because i2c1 will already be being used by LVGL

mysterywolf, 2022-08-07:
if you want to change to i2c2 please change here https://github.com/RT-Thread/rt-thread/blob/master/bsp/stm32/stm32f469-st-disco/applications/arduino_pinout/pins_arduino.h#L48

mysterywolf, 2022-08-07:
But, you actually can keep i2c1 and just use D14 and D15 like Arduino, because I2c is a bus.

neuberfran, 2022-08-07:
@mysterywolf I don't know if everything Ok yet. I don't have more folder user(s) afoter I copy SmartDrive.h and SmartDrive.cpp. I copy this files in two places

[image]
[image]

mysterywolf, 2022-08-07:
nop you need to copy the libraries with the folder or directly plug the zip file into the user file. rather two single files.

neuberfran, 2022-08-07:
@mysterywolf Please. Download the Zip file And show me (with images) What I need do. Will I have the user(s) folder again? Or will I have to create another project from scratch?

https://github.com/neuberfran/SmartDriveEmC

mysterywolf, 2022-08-07:
[image]
download the zip file and directly drag this zip file into user folder, you don't need to create a project again. After dragging the zip into user folder, you just need to click sync sconscript to project again.

neuberfran, 2022-08-07:
@mysterywolf  I have issues because Wire.h, etc...etc...

https://gist.github.com/neuberfran/15f9bc9ea6855f238da0ae3a4b72ebaa

[image]

neuberfran, 2022-08-07:
@mysterywolf 

I don't have smartdrive in i2c scan results yet. Despite everything connected and motor On

Do I need to create arduino.cpp and main.c files?
[image]

mysterywolf, 2022-08-07:
Please make sure your library is correct. This warnings comes from your library. Wire.h has been test through other arduino libraries successfully. 

Cearly, you didn't create the i2c2 bus device successfully. So arduino cannot use i2c2. I suggest you to use i2c1 bus, it has been initiated successfully.

mysterywolf, 2022-08-07:
> Do I need to create arduino.cpp and main.c files?

Nop , there is a arduino_main.cpp in application folder.

mysterywolf, 2022-08-07:
Serial.begin(); will be better rather than Serial.begin(9600);
Serial.begin(); means console serial will follow the rt-thread configurations which is 115200 by default.

neuberfran, 2022-08-07:
@mysterywolf I've been using i2c1 since the first moment. I still haven't been successful. Disregard what I showed about previous issues and Wire.h. I had put the SmartDrive.cpp in the wrong place(applications). I don't get compilation errors. I need to see how to debug or find the runtime error

mysterywolf, 2022-08-07:
Yap, please make sure your library is correct. Because Wrie.h has been tested under several arduino libraries.

neuberfran, 2022-08-07:
@mysterywolf 

Hey. With this code(arduino_main.cpp), I don't have more msh/> here. WhY? Because Serial.begin(9600)?

```

#include <Arduino.h>
#include <SmartDrive.h>

SmartDrive smd = SmartDrive(SmartDrive_DefaultAddress);

void setup(void)
{
    /* put your setup code here, to run once: */
/*  pinMode(LED_BUILTIN, OUTPUT); */
    Serial.begin(9600);
}

void loop(void)
{
    /* put your main code here, to run repeatedly:
    digitalWrite(LED_BUILTIN, !digitalRead(LED_BUILTIN));
    delay(100);*/

    Serial.print("Motor powered: " + smd.IsMotorPowered(SmartDrive_Motor_ID_1));
    delay(1000);
}

```

[image]

mysterywolf, 2022-08-07:
Yes, because RT-Thread default is 115200 and now you change to 9600.. of course it will not show any information if you keep using the terminal with 115200. So that I suggested you to use Serail.begin()

neuberfran, 2022-08-07:
@mysterywolf Motor powered: show nothing...

[image]

mysterywolf, 2022-08-07:
try to move SmartDrive() into setup function

neuberfran, 2022-08-07:
@mysterywolf Not Solved Yet. I will test in esp32 dev kit now. Although I already used this lib last week. Only it was on ubuntu 22.04/VsCode/PlatformIo

neuberfran, 2022-08-07:
@mysterywolf With esp32 I can run motor 1 with delay(5000);. I'm not put zip file. But I put smartdrive.cpp and smartdrive.h in folder Lib. I think..I stay using platformIo/ArduinoProject/VsCode

https://youtu.be/IEpiTdXtcsw

mysterywolf, 2022-08-07:
> I'm not put zip file. But I put smartdrive.cpp and smartdrive.h in folder Lib.

I fix this problem at here: https://github.com/RTduino/RTduino/commit/53dfa0aea0592ec70b1fc9fa61b9d9d677aa5227
> I think..I stay using platformIo/ArduinoProject/VsCode
Sorry for that. I cannot provide further support. Because I don't have the real hardwire or something like that. I have tested the Arduino UNO I2C libraries. Maybe still needs to be tested.

neuberfran, 2022-08-07:
@mysterywolf I'm pretty sure the problem is with regards to using i2c1. What do you say about it? Since I'm going to use i2c1 for lvgl

mysterywolf, 2022-08-07:
You may try to create a new project and only enable Arduino. Don't enable LVGL. Try it again.

neuberfran, 2022-08-07:
> I fix this problem at here

Ok. But inside de Zip folder have src folder and after have SmartDrive.h and SmartDrive.cpp

[image]

neuberfran, 2022-08-07:
> You may try to create a new project and only enable Arduino. Don't enable LVGL. Try it again.

Ok. I will do

mysterywolf, 2022-08-07:
IN this time, you can directly put these two single files in user folder, if you create a new project.

neuberfran, 2022-08-07:
@mysterywolf Not solved yet with Zip(Without LVGL). I change jumper wire from Vin to 5v. Because my external driver (SmartDrive) Work's with 5v. But nov solved

neuberfran, 2022-08-07:
> IN this time, you can directly put these two single files in user folder, if you create a new project.

I will try this

neuberfran, 2022-08-07:
@mysterywolf Sorry. I solve here. I have 2 i2c1 devices Now. 
I was connecting sda/scl on a5/a4 => pb8/pb9 and not on d15/d14. I connected to 5v

[image]

mysterywolf, 2022-08-07:
Ohhh
You need to read my reply and documents carefully...
[image]
[image]
[image]

closed 2022-08-07 by mysterywolf

neuberfran, 2022-08-09:
@mysterywolf I need tips, links, tutorials, videos on linking a design (photo below) made in LVGL with the above mentioned codes (main.c, arduino_main.cpp). That way I can give commands to my motor that is linked to the external i2c driver:

[image]

mysterywolf, 2022-08-09:
please go to LVGL community to ask this question.

www.lvgl.io

neuberfran, 2022-08-10:
@mysterywolf Pls. Help-me. I can't generate project.vcxproject Here. I have vs2019 and vs2022 community. Scons version is 3.0.5
[image]

mysterywolf, 2022-08-10:
https://youtu.be/k7QYk6hSwnc?t=1003

neuberfran, 2022-08-10:
@mysterywolf I stat trying this But I have issues in scons --target=vs2019(command). Look a last pic bro Pls

[46 more replies]