RT-Thread/rt-thread #6243: RT1170 SDRAM使能后，系统卡住

xjy198903, 2022-08-09:
在4.1.1版本中，rt1170-evk使能SDRAM，在主函数中使用rt_malloc函数分配内存，系统卡住，代码如下：
/*
 * Copyright (c) 2006-2022, RT-Thread Development Team
 *
 * SPDX-License-Identifier: Apache-2.0
 *
 * Change Logs:
 * Date           Author       Notes
 * 2019-05-06     tyustli      first version
 *
 */

#include <rtthread.h>
#include <rtdevice.h>
#include "drv_gpio.h"

char b_path[141072] = {1};

int main(void)
{
  while (1)
  {
    // rt_kprintf("test\r\n");
    char *full_path;
    full_path = (char *)rt_malloc(141072);
    rt_memcpy(full_path, b_path, 141072);
    rt_thread_mdelay(500);
  }
}

串口错误如下：
[image]

mysterywolf, 2022-08-09:
是否对full_path进行了检查? 是否为NULL？

mysterywolf, 2022-08-09:
你这么写肯定要死机的，没有对pull_path进行参数检查，而且还放在了循环里，1秒钟就要执行两次，过不了一会就会内存消耗光，然后就memset直接就死机了。

顺便用free命令看一下当前有多少内存

xjy198903, 2022-08-09:
注释掉申请内存的代码，可以看到：
[image]
，只要加了申请内存的代码：
#include <rtthread.h>
#include <rtdevice.h>
#include "drv_gpio.h"

char b_path[141072] = {1};

int main(void)
{
  // char *full_path;
  // full_path = (char *)rt_malloc(141072);
  // if (full_path != RT_NULL)
  // {
  //   rt_memcpy(full_path, b_path, 141072);
  // }

  while (1)
  {
    // rt_kprintf("test\r\n");

    rt_thread_mdelay(500);
  }
}
就是卡住，串口报错：
[image]

mysterywolf, 2022-08-09:
你把memcpy注释掉还死机吗？

xjy198903, 2022-08-09:
注释掉以后，不死机了，串口输出：
[image]
能看到它从系统HEAP里面分配空间了

mysterywolf, 2022-08-09:
[image]
把这个勾选上，再打开rt_memcpy试试？

xjy198903, 2022-08-09:
[image]
已加入选项，解除rt_memcpy后，仍还卡死

xjy198903, 2022-08-09:
补充说明：开发环境为IAR 9.30.1

mysterywolf, 2022-08-09:
昨天NXP官方已经帮你验证过了，我今天又确认了一次，他们那边可以正确运行，你的问题无法复现😂 1170支持IAR

mysterywolf, 2022-08-09:
[image]

mysterywolf, 2022-08-09:
[image]

xjy198903, 2022-08-09:
版本不一样，我用的是最新的4.1.1，而且官方的没说SDRAM有没有扩大到64M，目前BSP（drv_sdram.c和sdram_port.h）支持的是32M的大小

xjy198903, 2022-08-09:
切换到4.1.0版本，SDRAM一切正常，串口输出如下：
[image]
main函数操作如下：
#include <rtthread.h>
#include <rtdevice.h>
#include "drv_gpio.h"

int main(void)
{
    char *buf = (char*)rt_malloc(4*1024*1024);
    if(buf)
    {
      rt_memcpy(buf, "hello RT-Thread 4.1.0", 25);
      buf[26] = '\0';
    }
    rt_kprintf("buf: %s\r\n", buf);
    rt_free(buf);
    while (1)
    {
//      rt_kprintf("test\r\n");
      rt_thread_mdelay(500);
    }
}

xjy198903, 2022-08-09:
中间的数字是4*1024*1024，没有显示全

xjy198903, 2022-08-09:
信号显示不了，应该是`4*1024*1024`

mysterywolf, 2022-08-09:
https://github.com/RT-Thread/rt-thread/tree/master/bsp/imxrt/imxrt1170-nxp-evk
这个BSP是今年五月份才提交上来的，为什么能切到4.1.0版本呢，4.1.0是今年三月发布的呀

xjy198903, 2022-08-09:
我参照4.1.1，1170-nxp-evk文件夹拷贝过来，重复4.1.1版本的操作（引脚配置等等），重新编译烧录，就可以，可以肯定的是，4.1.1这部分有问题应该是kernel部分有待完善；上面的截图，NXP官方还玩的4.0.4，估计他们也是把现有代码拷贝去测试的

mysterywolf, 2022-08-09:
好的  这回应该清楚了 不是nxp的问题  我反馈一下争取4.1.1发布的时候解决

xjy198903, 2022-08-09:
嗯嗯，我在4.1.0版本上，再测试下1170别的接口☺

mysterywolf, 2022-08-09:
可否帮忙看一下Keil下是否也会发生同样的问题？ 4.1.1版本

xjy198903, 2022-08-09:
测试情况：Keil 5.30，使能SDRAM后，串口输出：
[image]
系统卡住

mysterywolf, 2022-08-09:
OK 感谢

mysterywolf, 2022-08-14:
HI, 你用目前的最新的代码，把这个commit revert回来之后试试 看看还死机嘛？
https://github.com/RT-Thread/rt-thread/commit/4562ae12460c22617185dd66332c5b591f256e7b

xjy198903, 2022-08-14:
你是说下载最新的代码，但是用原来的1170的bsp文件去验证吗？

mysterywolf, 2022-08-14:
不是 我怀疑是 这个提交记录导致的问题，请帮忙验证一下  https://github.com/RT-Thread/rt-thread/commit/4562ae12460c22617185dd66332c5b591f256e7b
你把这个提交记录里的内容恢复回来，也就是revert翻转回来，应该只翻转memheap.c的就可以
[image]
然后再测试一下rt_memcpy还死机否

记得用最新的代码

xjy198903, 2022-08-14:
嗯。清楚了，不过得明天了，我现在手上没板子

mysterywolf, 2022-08-14:
嗯嗯 谢谢

mysterywolf, 2022-08-14:
我在QEMU上测了一下 可以正确分配和memcpy 不会死机
我在怀疑 是不是你分配的内存不允许4字节拷贝还是有什么奇奇怪怪的限制

[image]

mysterywolf, 2022-08-14:
> 我在怀疑 是不是你分配的内存不允许4字节拷贝还是有什么奇奇怪怪的限制

不过也不应该 如果物理内存有限制的话 4.1.0也应该会崩

xjy198903, 2022-08-14:
那也应该是系统的问题，同样的代码，4.1.0自运行正确后，已连续运行1周无异常，同样的代码，在4.1.1立马死机

mysterywolf, 2022-08-14:
是的

xjy198903, 2022-08-14:
4.1.1如果直接用rt_memheap_alloc分配后，直接操作sdram可以随便操作，分配1兆没问题；4.1.0我现在用rt_malloc分配了4兆的空间，稳定运行

mysterywolf, 2022-08-14:
嗯 这肯定是内核的问题了 有时间先帮忙revert一下看看  我这边死活复现不出来

xjy198903, 2022-08-14:
好的，我尽快测试

xjy198903, 2022-08-15:
revert以后，注释掉ipc文件的RT_DEBUG_SCHEDULER_AVAILABLE(time != 0);增加SDRAM支持后，主函数如下：
/*
 * Copyright (c) 2006-2022, RT-Thread Development Team
 *
 * SPDX-License-Identifier: Apache-2.0
 *
 * Change Logs:
 * Date           Author       Notes
 * 2019-05-06     tyustli      first version
 *
 */

#include <rtthread.h>
#include <rtdevice.h>
#include "drv_gpio.h"

int main(void)
{
    char *buf = (char*)rt_malloc(4*1024*1024);
    if(buf)
    {
      rt_memcpy(buf, "hello RT-Thread 4.1.0 for RT1176", 39);
      buf[40] = '\0';
    }
    rt_kprintf("buf: %s\r\n", buf);
    rt_free(buf);
    while (1)
    {
//      rt_kprintf("test\r\n");
      rt_thread_mdelay(500);
    }
}
系统运行如下：

[image]

一切正常

mysterywolf, 2022-08-15:
好的 感谢！我来考虑怎么处理一下

xjy198903, 2022-08-15:
我要上传下SDRAM相关的配置，目前的BSP，SDRAM编译过不去，配置也不太对

mysterywolf, 2022-08-15:
嗯嗯好的
我看了一下这个commit里边的内容，这里边主要是检查memheap是否在调度器启动之后调用，因为memheap用到了互斥量和信号量，因此需要确保调度器启动之后才能够初始化memheap。是不是在这个bsp里边，有在调度器启动之前就开始初始化memheap了？

xjy198903, 2022-08-15:
rt_hw_board_init应该是在调度器起来之前就运行了，这里面有
#ifdef RT_USING_HEAP
    rt_system_heap_init((void *)HEAP_BEGIN, (void *)HEAP_END);
#endif

mysterywolf, 2022-08-15:
好的 问题点找到了 这里初始化有问题...

xjy198903, 2022-08-15:
我上传了sdram的修改，但是备注好像还是上一次的"RTL8211F驱动"相关的信息，merge的时候改下修改信息

mysterywolf, 2022-08-15:
没关系 这个我来处理一下即可

xjy198903, 2022-08-15:
SDRAM的修改还没有merge到master吧

mysterywolf, 2022-08-15:
merge了呀 你提交的PR我都合并了 你看一下是不是你上个PR漏提交了，或者是fork下来的rt-thread仓库没有同步rt-thread最新？

mysterywolf, 2022-08-15:
另外提交pr的时候建议基于你仓库的master新建一个分支来提交，不要直接往你仓库的master分支上提交。

xjy198903, 2022-08-15:
提示这个：
[image]
文件格式不对吗，还是什么的，我现在在master里面没看到这部分修改的内容

xjy198903, 2022-08-15:
> 另外提交pr的时候建议基于你仓库的master新建一个分支来提交，不要直接往你仓库的master分支上提交。

好的

mysterywolf, 2022-08-15:
应该是你全往master合并然后覆盖掉了 你看看可否重新提交一下？ 点击每一个update xxxx 这种记录就能看见你的当次提交内容

xjy198903, 2022-08-15:
好的，我fork后，新建分支再PR吧

mysterywolf, 2022-08-15:
提PR一般是这样：
从rt-thread/rt-thread仓库fork下来一份到你自己用户名下的rt-thread的仓库，然后，你用户名下的rt-thread仓库应该始终和rt-thread官方仓库的master分支保持同步，每次提交PR都从你仓库的master创建一个新的分支，一个分支一个目标比如修改网络或者什么的，然后基于此分支创建PR。然后可以创建下一个分支，再提另外一个主题的PR。

xjy198903, 2022-08-15:
正常开发来说是这样的，我这跨步骤了，为了省事

xjy198903, 2022-08-15:
已重新提交

mysterywolf, 2022-08-15:
请问你死机的时候第二个rt_memheap_init 是放在哪里初始化的？第一个rt_memheap_init就是你截图的位于hw_board_init函数里，第二个哩 您放在哪里了？

xjy198903, 2022-08-15:
第二个在drv_sdram.c里面，作为board init一部分进行的，你看一下

mysterywolf, 2022-08-15:
你把rt_hw_sdram_Init 塞到了 rt_hw_board_init里？

xjy198903, 2022-08-15:
不是我塞的，是原来bsp这么安排的，作为BOARD_INIT的一部分，初始化流程我没动过

mysterywolf, 2022-08-15:
噢噢 我知道问题在哪里了
我一会把你提交的那个PR改一下 然后你帮忙测试下
问题就出在这个BOARD_INIT上，我刚才以为你是直接塞到了rt_hw_board_init里，无论是那种方式，都会死机
第二个rt_memheap_init必须在调度器起来之后才能用

xjy198903, 2022-08-15:
好的，修改好了我试一下

mysterywolf, 2022-08-15:
合并进去了 你可以拉取一下最新的代码 试一试

xjy198903, 2022-08-15:
拉取最新代码，测试没问题，不过多了个编译警告，我把drv_sdram.h中init函数的声明注释掉了

[2 more replies]

closed 2022-08-15 by mysterywolf