RT-Thread/rt-thread #6260: How to integrate RT-Thread-i2c-LVGL project to LVGL design screen using RT-Thread studio Ide

neuberfran, 2022-08-12:
I have this menu(screen design) in **SLS** exported to **ui folder** of the directory below the image:
[image]
https://github.com/neuberfran/discorevy6_SquareLine_Project/tree/main/ui

[image]

https://forum.lvgl.io/t/how-to-integrate-rt-thread-i2c-lvgl-project-to-lvgl-design-screen-using-rt-thread-studio-ide/9550/5

**How to throw these files into RT-THREAD IDE/OS and make the function calls?**

mysterywolf, 2022-08-12:
You just put you application code, which generated from squareline studio, into lv_user_gui_init function.

neuberfran, 2022-08-12:
@mysterywolf 
```
int ui_init(void)
{
    rt_device_t device;

    device = rt_device_find("lcd");
    if(device == RT_NULL)
    {
       rt_kprintf("no graphic device in the system.\n");
       return-1;
     }
     /* re-set graphic device */
     rigui_graphic_set_device(device);

     {
                 externvoid application_init();
          application_init();
      }
      return0;
}
```
I pulled out the code above this link(page:90):
https://www.rt-thread.org/download/manual/study_rtt.pdf

As Mr Gabor Kiss-Vamosi said on the SLS forum I threw the ui_init files in the ui folder. How to use this function above to integrate my design?
(Since the music app didn't use this scheme of playing the files to ui folder)

NOTE: Now I have  **extern void lv_demo_neuber(void);**

[image]

neuberfran, 2022-08-12:
@mysterywolf 

**Issue:**
arm-none-eabi-gcc "../packages/lv_music_demo-v0.3.1/assets/img_lv_demo_rtt_music_btn_next.c"
arm-none-eabi-gcc "../packages/lv_music_demo-v0.3.1/assets/img_lv_demo_rtt_music_btn_pause.c"
../packages/ui/ui.c:18:6: error: #error "LV_COLOR_DEPTH should be 32bit to match SquareLine Studio's settings"
     #error "LV_COLOR_DEPTH should be 32bit to match SquareLine Studio's settings"
      ^
arm-none-eabi-gcc "../packages/lv_music_demo-v0.3.1/assets/img_lv_demo_rtt_music_btn_play.c"
make: *** [packages/ui/subdir.mk:21: packages/ui/ui.o] Error 1
make: *** Waiting for unfinished jobs....
"make -j8 all" terminated with exit code 2. Build might be incomplete.

https://github.com/neuberfran/discovery6_rtthreadPart2/blob/main/packages/ui/ui.c#L18

[image]

mysterywolf, 2022-08-12:
if you are using the squareline studio code, please disable the music player demo code.

neuberfran, 2022-08-12:
@mysterywolf 

#if LV_COLOR_DEPTH != 32 I change this line to #if LV_COLOR_DEPTH != 16
[image]
Finally. I have my first screen.
I need to call functions now.
**How to get started on this?**

mysterywolf, 2022-08-12:
> I need to call functions now.
> How to get started on this?

This is related LVGL application APIs. As rt-thread, a operating system community, cannot give your further imformation. I suggest you to read the LVGL documentation or use GOOGLE or youtube. There are many LVGL tutorials on internet.

closed 2022-08-14 by mysterywolf

neuberfran, 2022-08-14:
@mysterywolf 

Please,
https://gist.github.com/neuberfran/3c7bb51e029e49fa9ec5eede5ae10f0a
Look this ui.c in **///FUNCTIONS// Part**

**How to integrate ///FUNCTIONS/// with ///SCREENS////**?

///////////////////// FUNCTIONS ////////////////////
static void event_handler(lv_event_t * e)
{
    lv_event_code_t code = lv_event_get_code(e);
    if(code == LV_EVENT_CLICKED) {
        LV_LOG_USER("Clicked");
    }
    else if(code == LV_EVENT_VALUE_CHANGED) {
        LV_LOG_USER("Toggled");
    }
}
///////////////////// SCREENS ////////////////////
void ui_Screen1_screen_init(void)
{
    // ui_Screen1
    ui_Screen1 = lv_obj_create(NULL);
    lv_obj_clear_flag(ui_Screen1, LV_OBJ_FLAG_SCROLLABLE);
    // ui_rodarpdireita
    ui_rodarpdireita = lv_btn_create(ui_Screen1);
    lv_obj_set_width(ui_rodarpdireita, 125);
    lv_obj_set_height(ui_rodarpdireita, 50);
    lv_obj_set_x(ui_rodarpdireita, 8);
    lv_obj_set_y(ui_rodarpdireita, -47);
    lv_obj_set_align(ui_rodarpdireita, LV_ALIGN_CENTER);
    lv_obj_add_flag(ui_rodarpdireita, LV_OBJ_FLAG_SCROLL_ON_FOCUS);
    lv_obj_clear_flag(ui_rodarpdireita, LV_OBJ_FLAG_SCROLLABLE);