RT-Thread/rt-thread #6273: fatal error: perf_counter.h: No such file or directory compilation terminated.

neuberfran, 2022-08-14:
This issue insists on appearing in every project I start. In fact its does not appear right after the project is created, but after adding one or another package. LVGL type, etc.

Even if I remove the package, the error does not disappear and the project does not work again

[image]
[image]

mysterywolf, 2022-08-14:
You may go to the project folder and use ENV to download the pakcages `pkgs --update`

mysterywolf, 2022-08-14:
[image]

[image]

closed 2022-08-15 by mysterywolf

neuberfran, 2022-08-20:
@mysterywolf  I stay trying pkgs --update But this issue back again
[image]