RT-Thread/rt-thread #6276: Folder does not appear in packages: Permission issues or something wrong with RT-THREAD STUDIO IDE

neuberfran, 2022-08-14:
As is clear from the photos below, the folder does not appear where it should in the RT-THREAD STUDIO IDE. I've already done packages update and sync sconscript to project in RTT Studio and pksgs --update using ENV. It didn't.
It's not permission issue on my PC. There must be something wrong in the RTT Studio IDE.

[image]
[image]

mysterywolf, 2022-08-15:
This is not RT-Thread Studio problem.
ui folder is your application folder, you need to add it into your project by yourself. I suggest to move ui folder into application folder and modify the Sconscript file to add .c files and .h paths

neuberfran, 2022-08-15:
@mysterywolf 
Where be ui folder? I put in applications folder too. But nothing
[image]

neuberfran, 2022-08-15:
@mysterywolf 

In w11 (My Works Machine) I can't use simulator yet with vs2022.
I can't see lvgl (And Screen Music Demo) results list_thread:

scons --target=vs2012
scons: Reading SConscript files ...

scons: warning: No version of Visual Studio compiler found - C/C++ compilers most likely not set correctly
File "C:\test\rt-thread\bsp\simulator\SConstruct", line 13, in <module>

scons: warning: No version of Visual Studio compiler found - C/C++ compilers most likely not set correctly
File "C:\test\rt-thread\bsp\simulator\SConstruct", line 15, in <module>
**No vc version!**
scons: done reading SConscript files.
[image]
[image]

mysterywolf, 2022-08-15:
> @mysterywolf
> 
> In w11 (My Works Machine) I can't use simulator yet with vs2022. I can't see lvgl (And Screen Music Demo) results list_thread:
> 
> scons --target=vs2012 scons: Reading SConscript files ...
> 
> scons: warning: No version of Visual Studio compiler found - C/C++ compilers most likely not set correctly File "C:\test\rt-thread\bsp\simulator\SConstruct", line 13, in
> 
> scons: warning: No version of Visual Studio compiler found - C/C++ compilers most likely not set correctly File "C:\test\rt-thread\bsp\simulator\SConstruct", line 15, in **No vc version!** scons: done reading SConscript files. [image] [image]

You didn't enable LVGL, of course you cannot see LVGL demo

neuberfran, 2022-08-15:
> You didn't enable LVGL, of course you cannot see LVGL demo

> > @mysterywolf Where be ui folder? I put in applications folder too. But nothing [image]
> 
> You can copy this Sconscript into your folder. Then modify it, based on your C file position.

Everytime I need to import my old project from another Pc(Backup) to backup My ui folder. It's not Cool. How to configure SConscript file? Where will be SConscript file  that I will have to reconfigure?

[image]

neuberfran, 2022-08-15:
@mysterywolf 
Everytime I need to import my old project from another Pc(Backup) to backup My ui folder. It's not Cool. How to configure SConscript file? Where will be SConscript file that I will have to reconfigure?
[image]

neuberfran, 2022-08-15:
@mysterywolf 
Same issue using Env/scons command:

Look to ls -lha in ui folder. Not is wrong. I think

[image]

neuberfran, 2022-08-15:
> LVGL

I enable LVGL. But I can't see lvgl thread (and Music demo app screen). I think it's a windows 10 (and not windows 11) SDK

mysterywolf, 2022-08-15:
I submit a PR, you may try this https://github.com/lvgl/lvgl/commit/704363c5bdd5b29b00bf51807720f86c8f79bb6b

closed 2022-08-17 by mysterywolf