RT-Thread/rt-thread #6310: 中断通知的准确调用时机是什么？

miracleyfm, 2022-08-19:
中断通知的准确调用时机是什么？
  在一个中断服务程序里，如果要调用内核操作“rt_event_send/rt_mq_send”等，那必须在该中断服务程序开始调用rt_interrupt_enter，中断程序的结束位置调用rt_interrupt_leave？不这样做具体会导致操作系统调度出现什么样的紊乱后果？

miracleyfm, 2022-08-19:
目前我看了代码后，经过如下代码分析，其结论是：调用内核操作“rt_event_send/rt_mq_send”必须进行中断通知，否则会导致调度紊乱。
（1）rt_schedule利用了rt_interrupt_nest确定如何进行调度处理。而rt_interrupt_nest的值确定是当前是否处于中断状态，中断通知会设定rt_interrupt_nest的值；
（2）内核操作“rt_event_send、rt_mq_send_wait（被rt_mq_send调用）、rt_mb_send”（包括相应“接收”内核操作）都调用了rt_schedule()。

我是最近才开始深入学习RT_Thread的，我的理解是否正确，大家帮助下。

中断服务程序有内核操作时不进行所述的中断通知，系统紊乱到什么程度？我还没细致研究。应该是会引起系统功能不正常，包括信息丢失吧？

mysterywolf, 2022-08-19:
你好，请到论坛中提问问题，issue是用来反馈潜在bug 的 https://club.rt-thread.org/index.html

closed 2022-08-19 by mysterywolf

miracleyfm, 2022-10-11:
好的，非常感谢！_