RT-Thread/rt-thread #6335: support openconf for streamlining the macros in code

GuEe-GUI, 2022-08-26:
在现在 RT-Thread 的部分代码中，存在需要知道某个配置是否定义来作为一个代码运行的条件的用法：
```c
int main(void)
{
#ifdef RT_USING_ABC
    /* code 1 ... */
#else
    /* code 2 ... */
#endif
    return 0;
}
```
或者一些特殊商业项目要求禁止直接使用宏区分要执行的代码
```c
rt_bool_t has_abc =
#ifdef RT_USING_ABC
RT_TRUE
#else
RT_FALSE
#endif

int main(void)
{
    if (has_abc)
    {
        /* code 1 ... */
    }
    else
    {
        /* code 2 ... */
    }
    return 0;
}
```
如果类似的代码都这样写，一方面代码不是很美观。其次使用起来也麻烦，因此建议增加openconf的机制来作为代码执行的一部分来使用宏。

```c
/*
 * Helper macros to use CONFIG_ options in C/CPP expressions. Note that
 * these only work with boolean and tristate options.
 */

/*
 * Getting something that works in C and CPP for an arg that may or may
 * not be defined is tricky.  Here, if we have "#define CONFIG_BOOGER 1"
 * we match on the placeholder define, insert the "0," for arg1 and generate
 * the triplet (0, 1, 0).  Then the last step cherry picks the 2nd arg (a one).
 * When CONFIG_BOOGER is not defined, we generate a (... 1, 0) pair, and when
 * the last step cherry picks the 2nd arg, we get a zero.
 */
#define __ARG_PLACEHOLDER_1 0,
#define rt_config_enabled(cfg) _config_enabled(cfg)
#define _config_enabled(value) __config_enabled(__ARG_PLACEHOLDER_##value)
#define __config_enabled(arg1_or_junk) ___config_enabled(arg1_or_junk 1, 0)
#define ___config_enabled(__ignored, val, ...) val

/*
 * RT_IS_ENABLED(CONFIG_FOO) evaluates to 1 if CONFIG_FOO is set to 'y' or 'm',
 * 0 otherwise.
 *
 */
#define RT_IS_ENABLED(option) (rt_config_enabled(option) || rt_config_enabled(option##_MODULE))

/*
 * RT_IS_BUILTIN(CONFIG_FOO) evaluates to 1 if CONFIG_FOO is set to 'y', 0
 * otherwise. For boolean options, this is equivalent to
 * RT_IS_ENABLED(CONFIG_FOO).
 */
#define RT_IS_BUILTIN(option) rt_config_enabled(option)

/*
 * RT_IS_MODULE(CONFIG_FOO) evaluates to 1 if CONFIG_FOO is set to 'm', 0
 * otherwise.
 */
#define RT_IS_MODULE(option) rt_config_enabled(option##_MODULE)
```

使用案例：
```c
int main(void)
{
    if (RT_IS_ENABLED(RT_USING_ABC))
    {
        /* code 1 ... */
    }
    else
    {
        /* code 2 ... */
    }
    return 0;
}
```
这样子会节省一定量的代码，对于宏的使用也更加灵活。

BernardXiong, 2022-09-19:
这种还是把所有代码都编译进去的思路？宏太多确实变得超级麻烦了

GuEe-GUI, 2022-09-19:
> 

是的，不过编译器会把结果为`if (0)`或者`if (1) else`的分支优化掉。

BernardXiong, 2022-12-09:
> > 
> 
> 是的，不过编译器会把结果为`if (0)`或者`if (1) else`的分支优化掉。

这个方式似乎也是一种，不过这样代码展开就会出现代码不可达的情况了。也许可以找一些静态代码扫描软件，分析软件，也包括MISRA-C代码软件扫描下，看看是怎么样的。

closed 2026-09-02 by GuEe-GUI