RT-Thread/rt-thread #6339: lv_conf.h:11:0: error: unterminated #if. Enable arduino in LVGL. Issue in tutorial

neuberfran, 2022-08-28:
I stay trying this: https://daumemo.com/how-to-use-lvgl-library-on-arduino-with-an-esp-32-and-spi-lcd/#first-simple-lvgl-pro

in my stm32f469 disco kit. The first think to do in LVGL project part in #if 1 in lv_conf.h. But I can't. Why? **How to solve**
Pls

from ../packages/lv_music_demo-v0.3.1/assets/img_lv_demo_rtt_music_btn_prev.c:1:
C:\RT-ThreadStudio\workspace\discovery8\applications\lvgl/lv_conf.h:11:0: error: unterminated #if
 #if 1
 ^
make: *** [packages/lv_music_demo-v0.3.1/assets/subdir.mk:81: packages/lv_music_demo-v0.3.1/assets/img_lv_demo_rtt_music_btn_pause.o] Error 1
make: *** [packages/lv_music_demo-v0.3.1/assets/subdir.mk:81: packages/lv_music_demo-v0.3.1/assets/img_lv_demo_rtt_music_btn_play.o] Error 1
make: *** [packages/lv_music_demo-v0.3.1/assets/subdir.mk:81: packages/lv_music_demo-v0.3.1/assets/img_lv_demo_rtt_music_btn_prev.o] Error 1
"make -j8 all" terminated with exit code 2. Build might be incomplete.

[image]

mysterywolf, 2022-08-28:
https://www.techonthenet.com/c_language/directives/if.php

neuberfran, 2022-08-28:
@mysterywolf 

I'm trying this because I haven't been able to read a variable yet  inside a while <-> loop of RTduino:
**Maybe you know how to help me, because the motor only runs (more 5s) inside while <-> and I've already tried options**
https://stackoverflow.com/questions/73451177/how-to-implement-interrupt-or-task-inside-while-loop-in-cpp-file-without-voi
https://forum.lvgl.io/t/how-to-implement-interrupt-or-task-inside-while-loop-in-cpp-file-without-void-loop/9645/2
https://forum.arduino.cc/t/how-to-implement-interrupt-or-task-inside-while-loop-in-cpp-file-without-void-loop/1025352/9

neuberfran, 2022-08-28:
@mysterywolf Hey bro.
These issues mentioned above are very important for my project. Do you have anything to say about it?

mysterywolf, 2022-08-28:
If you don't want to use loop function. you can directly go to here: https://github.com/RTduino/RTduino/blob/master/core/arduino_thread.c#L81-L87

comment out the while loop and just use setup. in the setup function, you can organize your own while loop.

neuberfran, 2022-08-28:
@mysterywolf I think you misunderstood the issue. The problem has been getting the variable value inside a while <-> loop. Although I've already used delay(1000), it hasn't caught on yet. It is about getting the value asynchronously

mysterywolf, 2022-08-28:
ui runs in lvgl thread and the setup-loop runs in arduino thread, so you may need to use semaphore to sync it.

neuberfran, 2022-08-28:
@mysterywolf 
I saw this link: https://forum.squareline.io/t/how-can-i-use-mqtt-in-lvgl/153/5 just change ui_events.c to ui_events.cpp and everything was fine. If the motor rotates only 5s like in the video: https://www.youtube.com/watch?v=MAr3F34bJsk

I don't see why using void setup() and void loop() like in Arduino and I don't even think I need arduino_main.cpp. I think I can solve it in ui_events.cpp. What do you think? What is the solution?

Note: below you can see that I've tried several things in ui_events.cpp and it still didn't work.

`// SquareLine LVGL GENERATED FILE
// EDITOR VERSION: SquareLine Studio 1.0.5
// LVGL VERSION: 8.2
// PROJECT: discovery6

#include <Arduino.h>
#include "ui.h"
#include "vGlobal.h"
#include <SmartDrive.h>

SmartDrive smd = SmartDrive(SmartDrive_DefaultAddress);

volatile int buttonState = 0;

bool STOP01 =  true;

void run01right(lv_event_t * e)
{

// //   stopmotor01(lv_event_t * e);
//    ui_stop01 = lv_btn_create(ui_Screen1);
//  //  ui_stop02 = ui_event_stop01(lv_event_t * e);
//    delay( 1000 );
//  //    smd.Run_Seconds(SmartDrive_Motor_ID_1, SmartDrive_Dir_Reverse, 90, 25, STOP01, SmartDrive_Action_Brake );

//    STOP01 = true;

  //  smd.Run_Unlimited(SmartDrive_Motor_ID_1, SmartDrive_Dir_Reverse, 90);

//    ui_stop01 = lv_btn_create(ui_Screen1);

  //  lv_obj_add_event_cb(ui_stop01, ui_event_stop01, LV_EVENT_ALL, NULL);

  //  lv_event_code_t event = lv_event_get_code(e);
  //  lv_obj_t * ta = lv_event_get_target(e);
    while(STOP01) {
        smd.Run_Unlimited(SmartDrive_Motor_ID_1, SmartDrive_Dir_Reverse, 90);
        lv_event_code_t event = lv_event_get_code(e);
        lv_obj_t * ta = lv_event_get_target(e);
//        if (event == LV_EVENT_CLICKED) {
//            STOP01 = false;
//        }
        //lv_event_code_t event = lv_event_get_code(e);
       // lv_obj_t * ta = lv_event_get_target(e);
    }
}

void run02right(lv_event_t * e)
{
	// Your code here
    smd.Run_Unlimited(SmartDrive_Motor_ID_2, SmartDrive_Dir_Reverse, 90);
}

void stopmotor01(lv_event_t * e)
{
	// Your code here
    STOP01 = false;
    smd.StopMotor(SmartDrive_Motor_ID_1, SmartDrive_Action_Brake);
    STOP01 = false;
}

void run01left(lv_event_t * e)
{
    // Your code here
    while (STOP01)
    {
 //   stopmotor01(lv_event_t * e);
    smd.Run_Unlimited(SmartDrive_Motor_ID_1, SmartDrive_Dir_Forward, 90);
    delay( 1000 );
  //    smd.Run_Seconds(SmartDrive_Motor_ID_1, SmartDrive_Dir_Reverse, 90, 25, STOP01, SmartDrive_Action_Brake );
    }
    STOP01 = true;
}

void run02left(lv_event_t * e)
{
	// Your code here
    smd.Run_Unlimited(SmartDrive_Motor_ID_2, SmartDrive_Dir_Forward, 90);
}

void stopmotor02(lv_event_t * e)
{
	// Your code here
    smd.StopMotor(SmartDrive_Motor_ID_2, SmartDrive_Action_Brake);
}
`

mysterywolf, 2022-08-28:
Yes you can directly invoke ARDUINO API in any .cpp file or thread and get rid of setup-loop

neuberfran, 2022-08-28:
Please. What about LVGL_Arduino.ino in stm32f469 disco kit? How to use? How *ino file Can Help-me in this case?

2)If I use ui_events.cpp file Will I still need semaphore?

mysterywolf, 2022-08-29:
This is related your application layer code. I am not sure about that.

closed 2022-08-31 by neuberfran