RT-Thread/rt-thread #6351: no match for 'operator=='(operand types are 'lv_event_t*{aka_lv_event_t*}' and 'lv_event_code_t')

neuberfran, 2022-08-31:
Hi,

I stay trying integrante LVGL with RT-Thread OS/RTduino in stm32f469 disco kit. But I have issue:
**no match for 'operator=='(operand types are 'lv_event_t*{aka_lv_event_t*}' and 'lv_event_code_t')**
[image]

I need to know which button was pressed (ie inside ui_events.cpp). So, I need to get, in addition to the Click event, the Object (which button).

https://daumemo.com/how-to-use-lvgl-library-on-arduino-with-an-esp-32-and-spi-lcd/
I am basing myself on the site above, specifically on the part referred to in the photo below:
[image]

mysterywolf, 2022-08-31:
Are you sure your C language Grammar is correct? A pointer '==' to a number?

neuberfran, 2022-08-31:
@mysterywolf 
https://github.com/neuberfran/discovery9/blob/main/applications/lvgl/demo/ui/ui_events.cpp

neuberfran, 2022-09-01:
void run01right(lv_event_t * e, lv_obj_t * obj)
{
    while(STOP01) {
        smd.Run_Unlimited(SmartDrive_Motor_ID_1, SmartDrive_Dir_Reverse, 90);
      //  lv_event_code_t eb = lv_event_get_code(e);
        lv_event_code_t code = lv_event_get_code(e);
     //  lv_obj_t * obj = lv_event_get_target(obj);

  //      lv_obj_t * obj = lv_event_get_target(obj);

        if(code == LV_EVENT_CLICKED) {
              if (obj == ui_stop) {
                  smd.StopMotor(SmartDrive_Motor_ID_1, SmartDrive_Action_Brake);
              }
        }
    }
}

closed 2022-09-01 by neuberfran