RT-Thread/rt-thread #6368: rt-thread/ch32v307 NOT have ethernet connection AND RT-Thread terminal NOT have ifconfig command

neuberfran, 2022-09-02:
https://blog.csdn.net/bonson2004/article/details/122040352

I've been having success with the video tutorial above for two weeks now.

I enable UART1 to UART8. In ifconfig part I can see (using ifconfg e0 Not **ifconfig Only** egual tutorial) 192.168.1.30, etc, etc...But I can't ping to 192.168.1.2. I don't know why.

Evidently, I already tried with router, with switch And with crossover ethernet cable.

Why I don't have ifconfig command egual tutorial ?

How can i get ethernet/internet connection on ch32v307 board?

**Link about photos of this issue:**
https://drive.google.com/drive/folders/17KWvUCj2QosKkptQX4EfXS4R8O_JFvvH
```

**ifconfig command Only Work's with all arguments:**

msh />ifconfig e0
bad parameter! e.g: ifconfig e0 192.168.1.30 192.168.1.1 255.255.255.0

all arguments:
==============
msh />ifconfig e0 192.168.1.30 192.168.1.1 255.255.255.0
config : e0
IP addr: 192.168.1.30
Gateway: 192.168.1.1
netmask: 255.255.255.0
bad network interface device name(e0).
```
**My project with issue in GitHub:**
https://github.com/neuberfran/rtthread2/blob/main/applications/main.c

muaxiaohei, 2022-09-03:
In my test， I've used the BSP of 1.0.8 version and connected my ch32v307-r1-1v0 board with a router by a Network cable.
[image]
[image]

I've opened the ETH drivers which depends on the lwip assembly.
[image]
[image]
Here is my test result.
`msh >ping google.com`
`ping: from 172.217.160.110 icmp_seq=0 timeout`
`ping: from 172.217.160.110 icmp_seq=1 timeout`
`ping: from 172.217.160.110 icmp_seq=2 timeout`
`ping: from 172.217.160.110 icmp_seq=3 timeout`
[image]

maybe you should check more variables?
haha, excuse my poor English.

muaxiaohei, 2022-09-03:
I can't ping google.com success because I don't have a VPN.
I should paste the result of "ping baidu.com".

muaxiaohei, 2022-09-03:
And here is my test of ping a ip of LAN.
[image]

neuberfran, 2022-09-04:
@muaxiaohei Pls. Tell me ifconfig (Without Arguments) result. 

I'm learning to use this software:
https://www.big-meter.com/opensource/pt/6138a65ada0ebe0a421c38b3.html

[image]

muaxiaohei, 2022-09-04:
yep, here is the result.
[image]

neuberfran, 2022-09-04:
@muaxiaohei  Can u help-me in this issue?
https://stackoverflow.com/questions/73451177/how-to-implement-interrupt-or-task-inside-while-loop-in-cpp-file-without-voi?fbclid=IwAR1VSFMJRzDgd6dvsA8CCv7nKSLE8o5VkeZZO6UulEaDix66BLue4Vv6ocg
Any help is welcome on RT-Thread/Task/Thread/Semaphore/LVGL_Arduino/LVGL_Thread

muaxiaohei, 2022-09-05:
ooop, sorry, I am not familiar with LVGL, maybe later.

***@***.***
 
From: Neuber Sousa
Date: 2022-09-05 05:20
To: RT-Thread/rt-thread
CC: self-confident neko; Mention
Subject: Re: [RT-Thread/rt-thread] rt-thread/ch32v307 NOT have ethernet connection AND RT-Thread terminal NOT have ifconfig command (Issue #6368)
@muaxiaohei Can u help-me in this issue?
https://stackoverflow.com/questions/73451177/how-to-implement-interrupt-or-task-inside-while-loop-in-cpp-file-without-voi?fbclid=IwAR1VSFMJRzDgd6dvsA8CCv7nKSLE8o5VkeZZO6UulEaDix66BLue4Vv6ocg
Any help is welcome on RT-Thread/Task/Thread/Semaphore/LVGL_Arduino/LVGL_Thread

neuberfran, 2022-09-05:
@muaxiaohei  Pls. Where I can find this software to download for W11 or W10 Or in last case for Ubuntu 20.04?

[image]

muaxiaohei, 2022-09-05:
here is it.

***@***.***
 
From: Neuber Sousa
Date: 2022-09-05 09:02
To: RT-Thread/rt-thread
CC: self-confident neko; Mention
Subject: Re: [RT-Thread/rt-thread] rt-thread/ch32v307 NOT have ethernet connection AND RT-Thread terminal NOT have ifconfig command (Issue #6368)
@muaxiaohei Pls. Where I can find this software to download for W11 or W10 Or in last case for Ubuntu 20.04?

neuberfran, 2022-09-05:
@muaxiaohei Why did you post this? I asked politely and seriously

muaxiaohei, 2022-09-05:
> @muaxiaohei Why did you post this? I asked politely and seriously

I have added this software in my appendix， but it seems that github won't add this it. Here is my google driver sharing.

https://drive.google.com/file/d/14n0U9WYjKWwUXg0afXlzQUBmN44csFPZ/view?usp=sharing

neuberfran, 2022-09-05:
@muaxiaohei Why my folder examples\arduino\LVGL_Arduino appear in VsCode and Not appear in rt-thread studio ide? It's because SConscript? If yes, how to solve? How to solve

https://github.com/neuberfran/discovery9

[image]
[image]

muaxiaohei, 2022-09-06:
Maybe you should try to open the examples switch.

***@***.***
 
From: Neuber Sousa
Date: 2022-09-05 09:55
To: RT-Thread/rt-thread
CC: self-confident neko; Mention
Subject: Re: [RT-Thread/rt-thread] rt-thread/ch32v307 NOT have ethernet connection AND RT-Thread terminal NOT have ifconfig command (Issue #6368)
@muaxiaohei Why my folder examples\arduino\LVGL_Arduino appear in VsCode and Not appear in rt-thread studio ide? It's because SConscript? If yes, how to solve? How to solve
https://github.com/neuberfran/discovery9

neuberfran, 2022-09-06:
@muaxiaohei No. it's SConscript issue:
https://github.com/lvgl/lvgl/commit/704363c5bdd5b29b00bf51807720f86c8f79bb6b#diff-88fd1fd72a3632b24dc12027eb8b6a68bd13903191a70120caf5e57f3b76c03d
Cheers

closed 2022-09-11 by neuberfran