RT-Thread/rt-thread #6388: RT-Thread4.1.0工程用scons连续编译生成的map文件差异很大

childerxxx, 2022-09-07:
Dear RT-Thread developers,

一、编译环境：windows 和 Ubuntu20.4都能复现
二、软件版本：RT-Thread released v4.1.0, env_released_1.2.0,
三、复现路径：
1）下载RT-Thread v4.1.0源码
2）打开env_released_1.2.0, 进入文件夹 ..\rt-thread-4.1.0\bsp\stm32\stm32h750-armfly-h7-tool
3）输入`scons`进行编译，得到map文件，将map文件重命名为rtthread_1.map
4）输入`scons -c`清除上一次的编译结果，输入`scons`进行编译，得到map文件，将map文件重命名为rtthread_2.map
5）用文件对比工具对比rtthread_1.map和rtthread_2.map，即可看到巨大的差别。（.bin和.elf文件也有巨大的差别）

四、以下是一些补充描述：
Hi, 
我们在使用RT-Thread4.1.0的过程中，发现同样的源代码在scons编译后结果不同，这给我们带来了一些麻烦，因为**源代码相同，编译生成的bin文件却不同。**

刚开始所有编译生成的代码都能正常运行，后来我们遇到了一个版本的代码，**编译结果会影响运行结果。**
在编译出某个大小的hex的情况下可以正常运行，
在编译出另一大小的hex的编译结果下不能正常运行，
于是我们开始关注这个问题。

我们的代码连续编译得到的hex文件大小不同（连续编译5次就能复现了）
RT-Thread官方发布的源代码连续编译的hex大小也不同（可能要连续编译20次左右能够复现）
hex大小差异10~40个数字，但map和bin文件的差异超级大。

**有什么办法可以让它在源文件相同的情况下，每次编译生成bin文件都相同吗？**

4.1. 我们做过的尝试有：
1）改成单核编译，即scons -j8 改成scons -j1。结果：没有效果；
2）每次build前都清空上一次的编译结果，scons -c。结果：没有效果；

4.2. 下面是一些有助于您更好理解问题的截图
1）scons执行目录：
\rt-thread-4.1.0\bsp\stm32\stm32h750-armfly-h7-tool

2）编译生成文件对比图：
[image]
[image]

4.3. RT-Thread 4.1.0源码下载地址：
https://github.com/RT-Thread/rt-thread/releases/tag/v4.1.0

**我认为想要芯片运行行为一样，就必须先保证烧录进芯片的二进制.bin文件是一样的。这个理解是绝对正确的吗？也许二进制不一样，芯片的运行也能保持一致？**

[我在RT-Thread的论坛上也发布了提问](https://club.rt-thread.org/ask/question/d457e7c1e0aa43cd.html)：https://club.rt-thread.org/ask/question/d457e7c1e0aa43cd.html

欢迎留言，任何有关的想法或者思路都能帮助我们进一步深入地理解并解决这个问题，感谢！

Best regards,
Lan

mysterywolf, 2022-09-09:
这个PR https://github.com/RT-Thread/rt-thread/pull/6406 被错误合并了，CI检查这个PR导致了构建错误，可否重新提交一个PR来修复一下这个问题？
https://github.com/RT-Thread/rt-thread/runs/8268216170?check_suite_focus=true

childerxxx, 2022-09-09:
ok

mysterywolf, 2022-09-13:
感谢提交PR 此issue关闭