RT-Thread/rt-thread #6392: STM32 脉冲编码器驱动 自动重装载值设置问题

wdfk-prog, 2022-09-08:
硬件：STM32F429IGT6核心板
操作系统：RTT V4.1.5
使用设备驱动运行脉冲编码器。
运行情况：定时器10ms采集一次脉冲编码器的脉冲值并与上一次脉冲值做差，来得出速度
代码如下：【代码中的所有变量皆为int32_t 类型】

/**
  * @brief  编码器计数，速度计算
  * @param  None
  * @retval None
  * @note   https://blog.csdn.net/csol1607408930/article/details/112793292
            M法
*/
rt_int32_t walk_count;
rt_int32_t  walk_last_count;
static void Motor_Encoder_Calculation(void)
{
  /* 读取脉冲编码器计数值 */
  rt_device_read(pulse_encoder_dev, 0, & walk_count, 1);
  walk.motor.get_freq  = ( walk_count -  walk_last_count) * 10;
  walk.motor.get_speed = 60 * ( walk_count -  walk_last_count) * 10 / RESOLUTION;
  walk_last_count =  walk_count;
}
问题点：脉冲编码器过零点问题
使用设备框架运行脉冲编码器，读取的值只有正数。那么在过零点的时候，将会变为一个较大的整数。例如32755等等。这就要去处理过零点问题。很麻烦。
过零时断点截图如下

解决办法：将drv_pulse_encoder.c中的宏定义
#define AUTO_RELOAD_VALUE 0x7FFF
改为
#define AUTO_RELOAD_VALUE 0xFFFF
这样在过零点时，自动溢出，将变量值变成负数。这样做出来的速度就是正确的了。

应该有个设置，速度模式或者位置模式。速度模式用0XFFFF.角度模式用0X7FFF。这样就比较合适

bzrobotics, 2022-10-07:
驱动其实处理了过零问题，可以但是没有开溢出中断，开启这个后，溢出最大值是 int32_t 对于一般的小车是够了（65mm，900 脉冲一圈能跑 40 多公里），如果是 7 * 24 小时运行，用户层还是要处理溢出问题
可以看下#6488

wdfk-prog, 2022-10-08:
> 驱动其实处理了过零问题，没有生效是因为没有开溢出中断，开启这个后，溢出最大值是 int32_t 对于一般的小车是够了（65mm，900 脉冲一圈能跑 40 多公里），如果是 7 * 24 小时运行，用户层还是要处理溢出问题 可以看下#6488

多谢多谢，发现问题的时候看了下代码，看到溢出中断中是有处理的，就没感觉是代码问题了。
没想到是写了却没开中断...

closed 2022-10-11 by wdfk-prog