RT-Thread/rt-thread #6439: f1c100s and LVGL using RT-Thread

Mob1n, 2022-09-17:
Hello good friends of rtt.
the LVGL package, installed by rt-thread online packages in env needs alot of porting to do for the **allwinner TINA BSP**.
and also, there is a missing on **Lichee pi** BSP ; which is based on the same IC: **F1c100s** .

the biggest problem is of course the hardware packages are missing in rt-thread window . and i have to do it manually(from [here](https://gitee.com/jiang_xing/f1c100s_rt-thread) )  until the time comes and you guys do the humanity another favor and have it fully supported .
altogether , there is a huge need of f1c100s based BSPs in rt-thread softwares. studo or env.
so the question is : **How should I port the latest library of lvgl on f1c100s by rt-thread.**

rohanrehman, 2024-05-10:
Any progress on this?

Mob1n, 2024-05-10:
> Any progress on this?

as I recall from the time I did the project. the LVGL had a nice tutorial on how to port it for rt-thread devices. lookup porting on lvgl tutorial website.
as for the Allwinner's integration with rt-thread, still no progress. and I feel it never will have. so either change the MPU or look for older versions of rt-thread .