RT-Thread/rt-thread #6455: imxrt1020支持问题

guanjianhe, 2022-09-21:
打开`bsp/imxrt/imxrt1021-nxp-evk/`目录下的工程，编译失败错误，提示是启动文件有问题，报错信息如下所示：
```
..\libraries\MIMXRT1020\MIMXRT1021\arm\startup_MIMXRT1021.s(1): error: A1167E: Invalid line start 
..\libraries\MIMXRT1020\MIMXRT1021\arm\startup_MIMXRT1021.s(2): error: A1167E: Invalid line start 
..\libraries\MIMXRT1020\MIMXRT1021\arm\startup_MIMXRT1021.s(3): error: A1167E: Invalid line start 
..\libraries\MIMXRT1020\MIMXRT1021\arm\startup_MIMXRT1021.s(4): error: A1163E: Unknown opcode MIMXRT1021 , expecting opcode or Macro
```

mysterywolf, 2022-09-21:
使用env重新生成一下工程 scons --target=mdk5 或者 scons --target=iar

guanjianhe, 2022-09-23:
> 使用env重新生成一下工程 scons --target=mdk5 或者 scons --target=iar

不行，重新生成后还是报错误，错误如下所示：

```c
Build started: Project: project
*** Using Compiler 'V5.06 update 7 (build 960)', folder: 'D:\Keil\ARM\ARMCC\Bin'
Build target 'rtthread'
compiling cstring.c...
..\..\..\include\rtdef.h(400): error:  #20: identifier "RT_NAME_MAX" is undefined
      char       name[RT_NAME_MAX];                       /**< name of kernel object */
..\..\..\components\libc\compilers\common\cstring.c: 0 warnings, 1 error
compiling cstdio.c...
compiling cctype.c...
compiling syscall_mem.c...
..\..\..\include\rtdef.h(400): error:  #20: identifier "RT_NAME_MAX" is undefined
      char       name[RT_NAME_MAX];                       /**< name of kernel object */
..\..\..\components\libc\compilers\armlibc\syscall_mem.c: 0 warnings, 1 error
compiling syscalls.c...
..\..\..\include\rtdef.h(400): error:  #20: identifier "RT_NAME_MAX" is undefined
      char       name[RT_NAME_MAX];                       /**< name of kernel object */
..\..\..\components\libc\compilers\armlibc\syscalls.c: 0 warnings, 1 error
compiling main.c...
..\..\..\include\rtdef.h(400): error:  #20: identifier "RT_NAME_MAX" is undefined
      char       name[RT_NAME_MAX];                       /**< name of kernel object */
..\..\..\components\drivers\include\ipc/workqueue.h(41): error:  #70: incomplete type is not allowed
      struct rt_semaphore sem;
..\..\..\components\drivers\include\ipc/waitqueue.h(35): error:  #20: identifier "rt_wqueue_t" is undefined
  rt_inline void rt_wqueue_init(rt_wqueue_t *queue)
..\..\..\components\drivers\include\ipc/waitqueue.h(43): error:  #20: identifier "rt_wqueue_t" is undefined
  void rt_wqueue_add(rt_wqueue_t *queue, struct rt_wqueue_node *node);
..\..\..\components\drivers\include\ipc/waitqueue.h(45): error:  #20: identifier "rt_wqueue_t" is undefined
  int  rt_wqueue_wait(rt_wqueue_t *queue, int condition, int timeout);
..\..\..\components\drivers\include\ipc/waitqueue.h(46): error:  #20: identifier "rt_wqueue_t" is undefined
  void rt_wqueue_wakeup(rt_wqueue_t *queue, void *key);
..\..\..\components\drivers\include\ipc/pipe.h(21): error:  #70: incomplete type is not allowed
      struct rt_device parent;
..\..\..\components\drivers\include\ipc/pipe.h(31): error:  #20: identifier "rt_wqueue_t" is undefined
      rt_wqueue_t reader_queue;
..\..\..\components\drivers\include\ipc/pipe.h(32): error:  #20: identifier "rt_wqueue_t" is undefined
      rt_wqueue_t writer_queue;
..\..\..\components\drivers\include\ipc/pipe.h(34): error:  #70: incomplete type is not allowed
      struct rt_mutex lock;
..\..\..\components\drivers\include\ipc/poll.h(19): error:  #20: identifier "rt_wqueue_t" is undefined
  typedef void (*poll_queue_proc)(rt_wqueue_t *, struct rt_pollreq *);
..\..\..\components\drivers\include\ipc/poll.h(27): error:  #20: identifier "rt_wqueue_t" is undefined
  rt_inline void rt_poll_add(rt_wqueue_t *wq, rt_pollreq_t *req)
applications\main.c(22): warning:  #223-D: function "rt_pin_mode" declared implicitly
      rt_pin_mode(LED0_PIN, PIN_MODE_OUTPUT);
applications\main.c(22): error:  #20: identifier "PIN_MODE_OUTPUT" is undefined
      rt_pin_mode(LED0_PIN, PIN_MODE_OUTPUT);
applications\main.c(26): warning:  #223-D: function "rt_pin_write" declared implicitly
          rt_pin_write(LED0_PIN, PIN_HIGH);
applications\main.c(26): error:  #20: identifier "PIN_HIGH" is undefined
          rt_pin_write(LED0_PIN, PIN_HIGH);
applications\main.c(28): error:  #20: identifier "PIN_LOW" is undefined
          rt_pin_write(LED0_PIN, PIN_LOW);
applications\main.c: 2 warnings, 15 errors
compiling cstdlib.c...
..\..\..\include\rtdef.h(400): error:  #20: identifier "RT_NAME_MAX" is undefined
      char       name[RT_NAME_MAX];                       /**< name of kernel object */
..\..\..\components\libc\compilers\common\cstdlib.c: 0 warnings, 1 error
compiling div0.c...
compiling cwchar.c...
compiling showmem.c...
..\..\..\include\rtdef.h(400): error:  #20: identifier "RT_NAME_MAX" is undefined
      char       name[RT_NAME_MAX];                       /**< name of kernel object */
..\..\..\libcpu\arm\common\showmem.c: 0 warnings, 1 error
compiling backtrace.c...
..\..\..\include\rtdef.h(400): error:  #20: identifier "RT_NAME_MAX" is undefined
[... 2005 lines cut ...]
..\..\..\components\drivers\usb\usbdevice\core\usbdevice_core.c(1233): error:  #513: a value of type "int" cannot be assigned to an entity of type "void *"
      setting->desc = rt_malloc(desc_size);
..\..\..\components\drivers\usb\usbdevice\core\usbdevice_core.c(1237): warning:  #223-D: function "rt_free" declared implicitly
          rt_free(setting);
..\..\..\components\drivers\usb\usbdevice\core\usbdevice_core.c(1291): warning:  #223-D: function "rt_malloc" declared implicitly
      func = (ufunction_t)rt_malloc(sizeof(struct ufunction));
..\..\..\components\drivers\usb\usbdevice\core\usbdevice_core.c(1326): warning:  #223-D: function "rt_malloc" declared implicitly
      ep = (uep_t)rt_malloc(sizeof(struct uendpoint));
..\..\..\components\drivers\usb\usbdevice\core\usbdevice_core.c(2162): warning:  #223-D: function "rt_mq_recv" declared implicitly
          if(rt_mq_recv(&usb_mq, &msg, sizeof(struct udev_msg),
..\..\..\components\drivers\usb\usbdevice\core\usbdevice_core.c(2224): warning:  #223-D: function "rt_mq_send" declared implicitly
      return rt_mq_send(&usb_mq, (void*)msg, sizeof(struct udev_msg));
[... cut ...]
..\..\..\components\drivers\usb\usbdevice\core\usbdevice_core.c(2249): warning:  #223-D: function "rt_mq_init" declared implicitly
      rt_mq_init(&usb_mq,
..\..\..\components\drivers\usb\usbdevice\core\usbdevice_core.c(2144): error:  #235: variable "usb_mq" was declared with a never-completed type
  static struct rt_messagequeue usb_mq;
..\..\..\components\drivers\usb\usbdevice\core\usbdevice_core.c: 12 warnings, 5 errors
compiling fsl_flexspi_nor_boot.c...
compiling fsl_flexspi_nor_flash.c...
".\build\keil\Obj\rtthread.axf" - 750 Error(s), 242 Warning(s).
Target not created.
Build Time Elapsed:  00:00:26
```

cazure, 2022-10-02:
你应该是在bsp目录下直接 scons --target=mdk5的吧，
1.建议先复制bsp目录（bsp/imxrt/imxrt1021-nxp-evk 复制成 bsp/imxrt/imxrt1021-nxp-evk-test），然后更新在线软件包 pkgs --update，
2.再使用scons 直接调用gcc编译下，如果没有问题，
3.再scons --dist生成出dist目录下的工程（这个工程会把rtthread的源码和组件都复制到bsp包中，可以复制这个目录到任何文件夹中使用）
在这个工程里面再执行scons --target=mdk5 ，然后再使用mdk5编译。
因为你前面在bsp目录下直接转成mdk5工程好像mdk5里面rtthread的源码和组件都是用的相对路径 
像这样 ..\..\..\components\drivers\include
而你dist之后变成 rt-thread\components\drivers\include
可能是mdk5找路径没找对

guanjianhe, 2022-10-17:
> 你应该是在bsp目录下直接 scons --target=mdk5的吧， 1.建议先复制bsp目录（bsp/imxrt/imxrt1021-nxp-evk 复制成 bsp/imxrt/imxrt1021-nxp-evk-test），然后更新在线软件包 pkgs --update， 2.再使用scons 直接调用gcc编译下，如果没有问题， 3.再scons [...]

谢谢大佬，问题依旧存在