RT-Thread/rt-thread #6458: 🔨[action]添加action  CI 集成

supperthomas, 2022-09-22:
- [ ] 软件包添加aciton集成  参考https://club.rt-thread.org/ask/article/9683302d4d44f637.html
- [ ] 添加仓库检测qemu下所有软件包编译情况

supperthomas, 2022-11-25:
https://github.com/RT-Thread-packages/nimble

supperthomas, 2022-11-25:
https://github.com/supperthomas/RTTHREAD_SEGGER_TOOL

supperthomas, 2022-11-25:
https://github.com/xckhmf/nrfx

supperthomas, 2022-12-02:
https://github.com/supperthomas/LLSync_sdk_adapter

supperthomas, 2022-12-06:
如何给软件包加CI集成
https://club.rt-thread.org/ask/article/9683302d4d44f637.html

tuduweb, 2023-02-01:
*添加仓库检测qemu下所有软件包编译情况* 请问这个是什么意思呢？

supperthomas, 2023-02-01:
> _添加仓库检测qemu下所有软件包编译情况_ 请问这个是什么意思呢？

这个想测一测qemu可以用哪些软件包。把软件包通过CI 测一下编译情况

tuduweb, 2023-02-02:
> > _添加仓库检测qemu下所有软件包编译情况_ 请问这个是什么意思呢？
> 
> 这个想测一测qemu可以用哪些软件包。把软件包通过CI 测一下编译情况

这个听起来应该是官方仓库里，那一堆的`Jobs`集成到软件包里吧。也就是检查交叉编译是否能通过?

supperthomas, 2023-02-02:
> > > _添加仓库检测qemu下所有软件包编译情况_ 请问这个是什么意思呢？
> > 
> > 
> > 这个想测一测qemu可以用哪些软件包。把软件包通过CI 测一下编译情况
> 
> 这个听起来应该是官方仓库里，那一堆的`Jobs`集成到软件包里吧。也就是检查交叉编译是否能通过?

嗯，类似的。可以自己弄个仓库

supperthomas, 2023-02-02:
> > > > _添加仓库检测qemu下所有软件包编译情况_ 请问这个是什么意思呢？
> > > 
> > > 
> > > 这个想测一测qemu可以用哪些软件包。把软件包通过CI 测一下编译情况
> > 
> > 
> > 这个听起来应该是官方仓库里，那一堆的`Jobs`集成到软件包里吧。也就是检查交叉编译是否能通过?
> 
> 嗯，类似的。可以自己弄个仓库

这个qemu可能比较有点难度，可以先看上面那个。CI的。

tuduweb, 2023-02-02:
> > > > _添加仓库检测qemu下所有软件包编译情况_ 请问这个是什么意思呢？
> > > 
> > > 
> > > 这个想测一测qemu可以用哪些软件包。把软件包通过CI 测一下编译情况
> > 
> > 
> > 这个听起来应该是官方仓库里，那一堆的`Jobs`集成到软件包里吧。也就是检查交叉编译是否能通过?
> 
> 嗯，类似的。可以自己弄个仓库

我理解qemu应该是`qemu-vexpress-a9`, `qemu-virt64-aarch64`这些target?
那跟官方仓库action里普通的交叉编译，有什么本质上的区别呢?
是不是需要把对应的软件包repo给包含到官方镜像里，然后试一试能不能编译成功?

tuduweb, 2023-02-02:
> > > > > _添加仓库检测qemu下所有软件包编译情况_ 请问这个是什么意思呢？
> > > > 
> > > > 
> > > > 这个想测一测qemu可以用哪些软件包。把软件包通过CI 测一下编译情况
> > > 
> > > 
> > > 这个听起来应该是官方仓库里，那一堆的`Jobs`集成到软件包里吧。也就是检查交叉编译是否能通过?
> > 
> > 
> > 嗯，类似的。可以自己弄个仓库
> 
> 这个qemu可能比较有点难度，可以先看上面那个。CI的。

*软件包添加aciton集成*，哪里有这些需要添加CI的软件包呢？

supperthomas, 2023-02-02:
https://github.com/supperthomas/rtthread_software_package_list_show/blob/main/rtthread_softlist.md
@tuduweb 这是软件包列表：

具体怎么做可以参考：
https://github.com/supperthomas/RTTHREAD_SEGGER_TOOL
https://github.com/RT-Thread-packages/nimble

tuduweb, 2023-02-02:
> https://github.com/supperthomas/rtthread_software_package_list_show/blob/main/rtthread_softlist.md @tuduweb 这是软件包列表：
> 
> 具体怎么做可以参考： https://github.com/supperthomas/RTTHREAD_SEGGER_TOOL https://github.com/RT-Thread-packages/nimble

好的十分感谢!
qemu那部分，有时间的话，可以详细介绍一下大概是需要做一个什么样的任务。
最近正好做的东西涉及到了docker+qemu，有可能有些东西正好就对上了。

supperthomas, 2023-02-02:
> 好的十分感谢! qemu那部分，有时间的话，可以详细介绍一下大概是需要做一个什么样的任务。 最近正好做的东西涉及到了docker+qemu，有可能有些东西正好就对上了。

qemu:
添加一个仓库，专门用来测试qemu-vexpress-a9 这个平台下，所有软件包的编译情况，每次单独选择一个软件包，进行scons编译。主要为了查看，哪些软件包适合qemu下面编译，并且，通过qemu编译来检查软件包是否能和master上面的rt-thread代码编译正常。

zhkag, 2023-02-02:
官方现在有一个软件包的测试工具可以了解一下，这个测试工具也正在推进中

supperthomas, 2023-02-02:
> 官方现在有一个软件包的测试工具可以了解一下，这个测试工具也正在推进中

给个链接

zhkag, 2023-02-02:
> > 官方现在有一个软件包的测试工具可以了解一下，这个测试工具也正在推进中
> 
> 给个链接

https://github.com/RT-Thread/pkgs-test

tuduweb, 2023-02-02:
> > > https://github.com/supperthomas/rtthread_software_package_list_show/blob/main/rtthread_softlist.md @tuduweb 这是软件包列表：
> > > 具体怎么做可以参考： https://github.com/supperthomas/RTTHREAD_SEGGER_TOOL https://github.com/RT-Thread-packages/nimble
> > 
> > 
> > 好的十分感谢! qemu那部分，有时间的话，可以详细介绍一下大概是需要做一个什么样的任务。 最近正好做的东西涉及到了docker+qemu，有可能有些东西正好就对上了。
> 
> qemu: 添加一个仓库，专门用来测试qemu-vexpress-a9 这个平台下，所有软件包的编译情况，每次menuconfig单独选择一个软件包，进行scons编译。主要为了查看，哪些软件包适合qemu下面编译，并且，通过qemu编译来检查软件包是否能和master上面的rt-thread代码编译正常。
> 
> 建议先学习一下rtthread的qemu
> 
> 看下下面的链接。
> 
> https://www.rt-thread.org/document/site/#/rt-thread-version/rt-thread-standard/application-note/setup/qemu/ubuntu/an0005-qemu-ubuntu?id=%E5%9C%A8-ubuntu-%E5%B9%B3%E5%8F%B0%E5%BC%80%E5%8F%91-rt-thread
> 
> https://club.rt-thread.org/ask/article/d25fb0a88165f166.html

了解了，和master源一起结合起来编译看看能否通过。