RT-Thread/rt-thread #6463: DAC project to oscilloscope with issue: undefined reference to 'snif'. I have HANMATEK dos1102

neuberfran, 2022-09-22:
This is my project with issue:

https://github.com/neuberfran/CH32V307VCT6/blob/main/User/main.c

My project is based on this: https://github.com/lbthomsen/CH32V307V-EVT-R1/tree/master/oscillator
and this:
https://stm32world.com/wiki/CH32_Oscillator

**issue:**

                                                    ^~~~~~~~~
```
c:/mounriver2/mounriver_studio/toolchain/risc-v embedded gcc/bin/../lib/gcc/riscv-none-embed/8.2.0/../../../../riscv-none-embed/bin/ld.exe: ./Debug/debug.o: in function `_write':
C:\MRS_DATA\workspace\CH32V307VCT6\obj/../Debug/debug.c:152: multiple definition of `_write'; ./lib/debug.o:C:\MRS_DATA\workspace\CH32V307VCT6\obj/../lib/debug.c:90: first defined here
c:/mounriver2/mounriver_studio/toolchain/risc-v embedded gcc/bin/../lib/gcc/riscv-none-embed/8.2.0/../../../../riscv-none-embed/bin/ld.exe: ./Debug/debug.o: in function `_sbrk':
C:\MRS_DATA\workspace\CH32V307VCT6\obj/../Debug/debug.c:185: multiple definition of `_sbrk'; ./lib/debug.o:C:\MRS_DATA\workspace\CH32V307VCT6\obj/../lib/debug.c:123: first defined here
c:/mounriver2/mounriver_studio/toolchain/risc-v embedded gcc/bin/../lib/gcc/riscv-none-embed/8.2.0/../../../../riscv-none-embed/bin/ld.exe: ./User/main.o: in function `update_dac_buffer':
C:\MRS_DATA\workspace\CH32V307VCT6\obj/../User/main.c:36: undefined reference to `sinf'
c:/mounriver2/mounriver_studio/toolchain/risc-v embedded gcc/bin/../lib/gcc/riscv-none-embed/8.2.0/../../../../riscv-none-embed/bin/ld.exe: C:\MRS_DATA\workspace\CH32V307VCT6\obj/../User/main.c:37: undefined reference to `sinf'
collect2.exe: error: ld returned 1 exit status
make: *** [makefile:56: CH32V307VCT6.elf] Error 1
****
```

[image]

neuberfran, 2022-10-06:
@BernardXiong What? I don't understood

neuberfran, 2023-01-22:
https://github.com/RT-Thread/rt-thread/issues/6871

neuberfran, 2023-02-05:
https://github.com/RT-Thread/rt-thread/issues/6881

closed 2023-02-05 by neuberfran

igorschip, 2024-04-21:
It error is if sinf(var) bud if sinf([number]) error desapears.

Azerlag, 2025-03-18:
> It error is if sinf(var) bud if sinf([number]) error desapears.

You need to connect mathematical library (math.h) as -lm C Linker flag.
In the Libraries of Mounriver Studio (or Miscellaneous) section, find the Libraries (-l) field and add "m" there (this is -lm, but the IDE only specifies the library name without the -l prefix).