RT-Thread/rt-thread #6506: [bsp][stm32]CAN发送时拔插can总线，导致后续发送显示失败

wdfk-prog, 2022-10-11:
- 首先先连接can总线，正常发送数据成功后。
- 断开can总线，发送can数据，显示失败。这一步是正确的。
- 再次插回can总线，这时发送can数据，显示失败。
- 使用逻辑分析仪显示由ACK回复与can分析仪显示有具体数据
- 解决办法详细看帖
> https://club.rt-thread.org/ask/question/d728abb39aa22cb3.html
> https://club.rt-thread.org/ask/article/5cedb728813e6fd8.html
- 已在SMT32F429IGT6实验，标准CAN测试成功。CANOEPN测试成功

mysterywolf, 2022-10-11:
请问你用的是什么版本  你用最新的源码试试还有这个问题吗

wdfk-prog, 2022-10-12:
> 请问你用的是什么版本 你用最新的源码试试还有这个问题吗

版本忘记说明了，RTT使用V4.1.0。
drv_can.c和drv_can.h我有从最新的源码上复制过去。
components文件夹中的can.c是和源码一致的。
我发现了问题，就去找最新的can的代码运行了一下，并没有解决。

wdfk-prog, 2022-10-12:
> 请问你用的是什么版本 你用最新的源码试试还有这个问题吗

重新测试了一下，简单来说就是CAN被动错误恢复后，rt_device_write返回异常。
跟这个帖子描述的问题一样。
> https://club.rt-thread.org/ask/question/d19dd32f398cc2a6.html
看我发的帖子更直观。
> https://club.rt-thread.org/ask/question/d728abb39aa22cb3.html

wdfk-prog, 2022-10-12:
> 请问你用的是什么版本 你用最新的源码试试还有这个问题吗

# 主要是这个部分的代码的问题
- 能看一下是不是这个代码的问题。这是我看帖子别人的解决办法。测试了一下，确实能够解决
drv_can.c里，当CAN总线拔掉发送时，会进入ACK错误中断SCE，这时TSR的TXOK都是0，另CAN_FLAG_TXOK0/1/2值的定义，还要移位才能对应，READ_BIT没有进行移位，所以下面这三个条件并不会执行，改为下面方式就可以执行了，并且清理了发送请求位CAN_TSR_RQCP0/1/2
```c
CAN1_SCE_IRQHandler(void)
{
    ......
    case RT_CAN_BUS_ACK_ERR:/* attention !!! test ack err's unit is transmit unit */
        drv_can1.device.status.ackerrcnt++;
        if (!READ_BIT(drv_can1.CanHandle.Instance->TSR, CAN_FLAG_TXOK0))
            rt_hw_can_isr(&drv_can1.device, RT_CAN_EVENT_TX_FAIL | 0 << 8);
        else if (!READ_BIT(drv_can1.CanHandle.Instance->TSR, CAN_FLAG_TXOK1))
            rt_hw_can_isr(&drv_can1.device, RT_CAN_EVENT_TX_FAIL | 1 << 8);
        else if (!READ_BIT(drv_can1.CanHandle.Instance->TSR, CAN_FLAG_TXOK2))
            rt_hw_can_isr(&drv_can1.device, RT_CAN_EVENT_TX_FAIL | 2 << 8);
        break;
    ......
}
```
———————-改为以下代码———————

```c
CAN1_SCE_IRQHandler(void)
{
    ......
    case RT_CAN_BUS_ACK_ERR:/* attention !!! test ack err's unit is transmit unit */
        drv_can1.device.status.ackerrcnt++;
        if (__HAL_CAN_GET_FLAG(hcan, CAN_FLAG_RQCP0))
        {
            if (!__HAL_CAN_GET_FLAG(hcan, CAN_FLAG_TXOK0))
            {
                rt_hw_can_isr(&drv_can1.device, RT_CAN_EVENT_TX_FAIL | 0 << 8);
            }
            SET_BIT(hcan->Instance->TSR, CAN_TSR_RQCP0);
        }
        else if (__HAL_CAN_GET_FLAG(hcan, CAN_FLAG_RQCP1))
        {
            if (!__HAL_CAN_GET_FLAG(hcan, CAN_FLAG_TXOK1))
            {
                rt_hw_can_isr(&drv_can1.device, RT_CAN_EVENT_TX_FAIL | 1 << 8);
            }
            SET_BIT(hcan->Instance->TSR, CAN_TSR_RQCP1);
        }
        else if (__HAL_CAN_GET_FLAG(hcan, CAN_FLAG_RQCP2))
        {
            if (!__HAL_CAN_GET_FLAG(hcan, CAN_FLAG_TXOK2))
            {
                rt_hw_can_isr(&drv_can1.device, RT_CAN_EVENT_TX_FAIL | 2 << 8);
            }
            SET_BIT(hcan->Instance->TSR, CAN_TSR_RQCP2);
        }
        break;
    ......
}
```

mysterywolf, 2022-10-12:
可否提交一个PR~

wdfk-prog, 2022-10-12:
> 可否提交一个PR~
已提交
https://github.com/RT-Thread/rt-thread/pull/6511

mysterywolf, 2022-10-12:
感谢~

wdfk-prog, 2022-10-17:
使用自动重传模式下，位显性错误发生后，阻塞当前线程。(短路CAN总线)

closed 2022-10-24 by wdfk-prog