RT-Thread/rt-thread #6565: 几处未知原因的rtthread utest问题

jifanchn, 2022-10-27:
我使用GD32F103RCT6开发板移植STM32的相关BSP，在UTEST中遇到几个问题，其中部分已提交PR，另外几处确实不熟悉原因：

1. mailbox_tc.c中，thread3_recv_dynamic_mb，result = rt_mb_recv(test_dynamic_mb, (rt_ubase_t *)&mb_recv_str2, RT_WAITING_FOREVER);，会在第二次运行utest时失败，打断点发现在ipc.c:2318行，rt_schedule();行失败，然后线程返回错误。
2. thread_tc.c中，test_thread_suspend函数，会在此test case中，使用rt_thread_suspend中断其它线程，但看上去rt_thread_suspend是不允许其它线程中断的，因RT_ASSERT(thread == rt_thread_self());约束

还请大佬们指教

jifanchn, 2022-10-27:
commit id: fdcee2da209fa587260d6b92bb37bb721bc6e5d5

mysterywolf, 2022-10-27:
是不是一部分utest根本没有想到会被执行第二次，我在example里也发现了这个问题，有些变量、资源没有回收或者置零，根本没考虑到下次执行的问题。  @Guozhanxin

jifanchn, 2022-10-30:
thread_tc.c这里在线程1中调用rt_thread_suspend挂起线程2，是否是test case的问题？

对于rtthread内核还是不甚熟悉，如果确认的话我可以提交PR，尝试改动这个test case。
但是也相心这个test case是经过测试验证的，是否某早期版本允许这样做，或者有什么历史原因？

mysterywolf, 2022-10-31:
你好，确实是testcase有问题，rt-thread的线程挂起函数限制比较严格，不允许一个线程挂起另一个线程，正在商量如何解决。

yangjie11, 2023-03-24:
1.mailbox tc 第二次测试失败     
[image]

2.thread tc 中，对test_thread_suspend函数的测试已处于屏蔽状态

closed 2026-11-03 by Rbb666