RT-Thread/rt-thread #6567: ULOG能否支持分开输出到文件后端和控制台，而不是一起输出

wdfk-prog, 2022-10-27:
能否支持分开输出。

-   一、例如这个文章所实现的。
> https://club.rt-thread.org/ask/article/e3ace649aea0ba5c.html
```c
enum _LOG_ID
{
    LOG_SYS_ID = 0x00,//系统日志ID，可以输出到控制台显示
    LOG_PM_ID,//文件后端ID，只需要输出到文件后端，不需要打印到控制台中
   。。。//输出到其他地方
    LOG_MAX_ID,
};
//例如增加一个ID选择，分开不同的输出选择。
ulog_output(LOG_ID, LOG_LVL_DBG, TAG, RT_TRUE, __VA_ARGS__)
```

因为有些日志是需要不断输出的，但是调试的时候又不需要关心这些日志。
例如小车的运动参数，平常我不关心他怎么动。但是出问题的时候，又需要看看日志咋回事。

另外增加一个可以打开输出到控制台的选项，就可以调试时需要时显示，不需要时关闭。

- 二、另外想法是日志过滤功能的实现更改一下
之前想实现这功能，想到了日志过滤。
但是在代码内日志过滤了，就啥也不输出了。文件后端也没有，控制台也没有显示。
如果滤波功能可以更改一下，支持文件后端和控制台分开滤波也是可以的

armink, 2022-10-27:
> 因为有些日志是需要不断输出的，但是调试的时候又不需要关心这些日志。
> 例如小车的运动参数，平常我不关心他怎么动。但是出问题的时候，又需要看看日志咋回事

可以使用 API ulog_tag_lvl_filter_set 按日志标签，动态设置输出级别，或者使用 ulog_tag_lvl 命令动态设置

> 但是在代码内日志过滤了，就啥也不输出了。文件后端也没有，控制台也没有显示。
> 如果滤波功能可以更改一下，支持文件后端和控制台分开滤波也是可以的

每个后端，比如控制台或者文件，都可以设置其独立的后端过滤器，详见 API ulog_backend_set_filter

armink, 2022-10-27:
搞定后，期待分享一个应用笔记哈

wdfk-prog, 2022-10-28:
> 搞定后，期待分享一个应用笔记哈

已实现。Thanks♪(･ω･)ﾉ

closed 2022-10-28 by wdfk-prog