RT-Thread/rt-thread #6568: 文档中心设备驱动DAC的示例函数输入有错误

eternal-echo, 2022-10-28:
示例中
```c
    /* 设置输出值 */
    value = atoi(argv[1]);
    rt_dac_write(dac_dev, DAC_DEV_NAME, DAC_DEV_CHANNEL, &value);
    rt_kprintf("the value is :%d \n", value);
```
函数原型为
```c
rt_uint32_t rt_dac_write(rt_dac_device_t dev, rt_uint32_t channel, rt_uint32_t value)
{
    RT_ASSERT(dev);

    dev->ops->convert(dev, channel, &value);

    return RT_EOK;
}
```
第三个参数应去掉&符号，应改为：
```c
    /* 设置输出值 */
    value = atoi(argv[1]);
    rt_dac_write(dac_dev, DAC_DEV_NAME, DAC_DEV_CHANNEL, value);
    rt_kprintf("the value is :%d \n", value);
```

KemGong, 2023-03-17:
在 “文档中心设备驱动中似乎没有找到DAC的示例

CXSforHPU, 2026-10-13:
* 该issu中，已合并修复。
* 文档中心参数描述也已经更新，但文档格式存在问题如下

[image]
* 针对文档格式解决提出一下解决方案

* master
    * https://gitee.com/rtthread/docs-online/pulls/737
* v5.2.x
    * https://gitee.com/rtthread/docs-online/pulls/738
* v5.1.x
    * https://gitee.com/rtthread/docs-online/pulls/739
* v5.0.x
    * https://gitee.com/rtthread/docs-online/pulls/740

建议关闭该issue