RT-Thread/rt-thread #6587: cmd.c中list_thread等函数printf格式“%-*.s”是否有问题

wdfk-prog, 2022-11-03:
### list_thread()函数中的printf有 .precision（精度）
```c
rt_kprintf("%-*.s pri  status      sp     stack size max used left tick  error\n", maxlen, item_title);
```
### C关于 .precision（精度）描述
- 对于 s: 要输出的最大字符数。默认情况下，所有字符都会被输出，直到遇到末尾的空字符。
[image]
### 没有说明“.s”的情况，这种情况下，不同printf的处理结果不一样
- 用kservice.c的rt_vsnprintf可以输出thread"
[image]
- 用rt_vsnprintf_full软件包中的rt_vsnprintf输出不了“thread”
>https://github.com/mysterywolf/rt_vsnprintf_full
[image]
- 使用printf输出结果
[image]
### kservice.c中相关代码如下
```c
       while (1)
        {
            /* skips the first '%' also */
            ++ fmt;
            if (*fmt == '-') flags |= LEFT;
            else if (*fmt == '+') flags |= PLUS;
            else if (*fmt == ' .') flags |= SPACE;
            else if (*fmt == '#') flags |= SPECIAL;
            else if (*fmt == '0') flags |= ZEROPAD;
            else break;
        }

        /* get field width */
        field_width = -1;
        if (_ISDIGIT(*fmt)) field_width = skip_atoi(&fmt);
        else if (*fmt == '*')
        {
            ++ fmt;
            /* it's the next argument */
            field_width = va_arg(args, int);
            if (field_width < 0)
            {
                field_width = -field_width;
                flags |= LEFT;
            }
        }
```
### rt_vsnprintf_full中相关代码如下
```c
    // evaluate flags
    flags = 0U;
    do {
      switch (*format) {
        case '0': flags |= FLAGS_ZEROPAD; format++; n = 1U; break;
        case '-': flags |= FLAGS_LEFT;    format++; n = 1U; break;
        case '+': flags |= FLAGS_PLUS;    format++; n = 1U; break;
        case ' ': flags |= FLAGS_SPACE;   format++; n = 1U; break;
        case '#': flags |= FLAGS_HASH;    format++; n = 1U; break;
        default :                                   n = 0U; break;
      }
    } while (n);

    // evaluate width field
    width = 0U;
    if (is_digit_(*format)) {
      width = atoi_(&format);
    }
    else if (*format == '*') {
      const int w = va_arg(va, int);
      if (w < 0) {
        flags |= FLAGS_LEFT;    // reverse padding
        width = (unsigned int)-w;
      }
      else {
        width = (unsigned int)w;
      }
      format++;
    }

    // evaluate precision field
    precision = 0U;
    if (*format == '.') {
      flags |= FLAGS_PRECISION;
      format++;
      if (is_digit_(*format)) {
        precision = atoi_(&format);
      }
      else if (*format == '*') {
        const int precision_ = (int)va_arg(va, int);
        precision = precision_ > 0 ? (unsigned int)precision_ : 0U;
        format++;
      }
    }
```
### 是否应该规范格式使用“%-*.*s”
如下
```c
 rt_kprintf("%-*.*s pri  status      sp     stack size max used left tick  error\n", maxlen, maxlen,item_title);
```
- 用kservice.c的rt_vsnprintf可以输出thread"
[image]

- 用rt_vsnprintf_full软件包中的rt_vsnprintf输出结果
[image]

- printf输出结果
[image]

mysterywolf, 2022-11-03:
你好 麻烦随便往我邮箱里发点什么

mysterywolf, 2022-12-15:
这个部分有时间可否把整个仓库扫描检查一下 有问题的提个PR改一下 按照标准的格式走~

mysterywolf, 2022-12-15:
不用管原生rt_kprintf，只要开了rt_vsnprintf_full这个软件包能正确显示就可以

closed 2022-12-23 by mysterywolf