RT-Thread/rt-thread #6588: 使用 gcc10 编译，找不到 fd_mask

yuqingli05, 2022-11-03:
使用env编译，工具链使用 10.3.1 20210824 (release)，提示找不到 'fd_mask' 。使用env自带工具链正常
F:\folder\rtt\rt-thread\components\libc\posix\io\poll\select.c:24:10: error: 'fd_mask' undeclared (first use in this function)

mysterywolf, 2022-11-04:
这是哪个rt-thread版本啊？

yuqingli05, 2022-11-04:
> 这是哪个rt-thread版本啊？
4.1.0
SHA-1: 201bd8871dff1efd1f48bdfd01a1938e4547ffd5

mysterywolf, 2022-11-04:
你好 请使用4.1.1版本，4.1.0版本有这个问题 但是在4.1.1已经修复了
https://github.com/RT-Thread/rt-thread/pull/5790/files#diff-e4b675da48f427ed37d5a1417bec17b7c82e849bcd64610db740fec491bc5a53

yuqingli05, 2022-11-04:
> 你好 请使用4.1.1版本，4.1.0版本有这个问题 但是在4.1.1已经修复了 https://github.com/RT-Thread/rt-thread/pull/5790/files#diff-e4b675da48f427ed37d5a1417bec17b7c82e849bcd64610db740fec491bc5a53

用的 gitee的 仓库，没看到 4.1.1 的标签

mysterywolf, 2022-11-04:
你直接用gitee 切到master分支吧 注意是master分支 不是默认的gitee_master分支

closed 2022-11-04 by mysterywolf