RT-Thread/rt-thread #6609: can-not-build-for-nuclei

openluopworld, 2022-11-11:
RT Thread Version: 4.1.0<br>
RT Thread env: 从官网下载的1.2.0版本<br>
Nuclei Toolchain: Nuclei Studio 202102版本自带的工具链<br>
Nuclei SDK: release 0.3.9<br>

步骤：<br>
1、在环境变量中配置好nuclei toolchain中gcc和openocd的路径<br>
2、按照rt-thread\bsp\nuclei\gd32vf103_rvstar的说明，在rt-thread\bsp\nuclei\gd32vf103_rvstar目录下启动env<br>
3、执行pkgs --update时报错<br>
```sh
> pkgs --update                                                                                                                
Traceback (most recent call last):                                                                                             
  File "D:\101-software-install\env_released_1.2.0\env\tools\ConEmu\ConEmu\..\..\..\tools\scripts\env.py", line 94, in <module>
                                                                                                                               
    main()                                                                                                                     
  File "D:\101-software-install\env_released_1.2.0\env\tools\ConEmu\ConEmu\..\..\..\tools\scripts\env.py", line 90, in main    
    args.func(args)                                                                                                            
  File "D:\101-software-install\env_released_1.2.0\env\tools\scripts\cmds\cmd_package.py", line 1122, in cmd                   
    package_update()                                                                                                           
  File "D:\101-software-install\env_released_1.2.0\env\tools\scripts\cmds\cmd_package.py", line 874, in package_update         
    if install_pkg(env_root, pkgs_root, bsp_root, pkg):                                                                        
  File "D:\101-software-install\env_released_1.2.0\env\tools\scripts\cmds\cmd_package.py", line 271, in install_pkg            
    package.parse(pkg_path)                                                                                                    
  File "D:\101-software-install\env_released_1.2.0\env\tools\scripts\package.py", line 135, in parse                           
    with open(filename, "r") as f:                                                                                             
IOError: [Errno 2] No such file or directory: 'D:\\101-software-install\\env_released_1.2.0\\env\\tools\\ConEmu\\ConEmu\\..\\..
\\..\\packages\\packages/peripherals/nuclei_sdk\\package.json'                                                                 
```

openluopworld, 2022-11-11:
我注意到这里有相关的配置，https://github.com/RT-Thread/packages/tree/master/peripherals/nuclei_sdk，先试试。

openluopworld, 2022-11-11:
当前下载sdk后编译提示工具链找不到，步骤如下：

1、手动下载nuclei-sdk-0.3.9，并放置到rt-thread\bsp\nuclei\gd32vf103_rvstar\packages路径下

2、在rt-thread\bsp\nuclei\gd32vf103_rvstar\packages下创建setup_config.bat，并添加如下内容
```sh
set NUCLEI_TOOL_ROOT=D:\100-software\NucleiStudio\toolchain
```

3、在rt-thread\bsp\nuclei\gd32vf103_rvstar目录下打开env工具，执行rt-thread\bsp\nuclei\gd32vf103_rvstar\packages下的setup.bat脚本

4、运行scons -c提示如下
```sh
> scons -c
scons: Reading SConscript files ...
Supported downloaded modes for board gd32vf103v_rvstar are flashxip, chosen downloaded mode is flashxip
Newlib version:unknown
Error: the toolchain path (D:/Software/Nuclei/gcc/bin) is not exist, please check 'EXEC_PATH' in path or rtconfig.py.
```

5、将rtconfig.py脚本的EXEC_PATH修改为真实的路径D:/100-software/NucleiStudio/toolchain/gcc/bin，执行scons -c成功，<b>执行scons提示工具链找不到</b>
```sh
> scons -c                                                                                                                     
scons: Reading SConscript files ...                                                                                            
Supported downloaded modes for board gd32vf103v_rvstar are flashxip, chosen downloaded mode is flashxip                        
Newlib version:3.0.0                                                                                                           
scons: done reading SConscript files.                                                                                          
scons: Cleaning targets ...                                                                                                    
Removed cconfig.h                                                                                                              
scons: done cleaning targets.                                                                                                  
                                                                                                                               
luopeng@DESKTOP-HSS3LCM E:\ctexdoc\reqtex\rt-thread\bsp\nuclei\gd32vf103_rvstar                                                
>                                                                                                                              
luopeng@DESKTOP-HSS3LCM E:\ctexdoc\reqtex\rt-thread\bsp\nuclei\gd32vf103_rvstar                                                
> scons                                                                                                                        
scons: Reading SConscript files ...                                                                                            
Supported downloaded modes for board gd32vf103v_rvstar are flashxip, chosen downloaded mode is flashxip                        
Newlib version:3.0.0                                                                                                           
scons: done reading SConscript files.                                                                                          
scons: Building targets ...                                                                                                    
scons: building associated VariantDir targets: build                                                                           
CC build\applications\main.o                                                                                                   
riscv-nuclei-elf-gcc: error:  -march=rv32imac -mabi=ilp32 -mcmodel=medany : No such file or directory                          
scons: *** [build\applications\main.o] Error 1                                                                                 
scons: building terminated because of errors.                                                                                  
                                                                                                                               
luopeng@DESKTOP-HSS3LCM E:\ctexdoc\reqtex\rt-thread\bsp\nuclei\gd32vf103_rvstar                                                
> which riscv-nuclei-elf-gcc                                                                                                   
'which' is not recognized as an internal or external command,                                                                  
operable program or batch file.                                                                                                
                                                                                                                               
luopeng@DESKTOP-HSS3LCM E:\ctexdoc\reqtex\rt-thread\bsp\nuclei\gd32vf103_rvstar                                                
> riscv-nuclei-elf-gcc -v                                                                                                      
Using built-in specs.                                                                                                          
COLLECT_GCC=riscv-nuclei-elf-gcc                                                                                               
COLLECT_LTO_WRAPPER=d:/100-software/nucleistudio/toolchain/gcc/bin/../libexec/gcc/riscv-nuclei-elf/9.2.0/lto-wrapper.exe       
Target: riscv-nuclei-elf                                                                                                       
Configured with: /home/xl_ci/toolchain/riscv-gnu-toolchain/riscv-gcc/configure --target=riscv-nuclei-elf --host=i686-w64-mingw3
[... cut ...]
gomp --disable-nls --src=/home/xl_ci/toolchain/riscv-gnu-toolchain/riscv-gcc --enable-checking=yes --enable-multilib --with-abi
=ilp32d --with-arch=rv32gc --with-tune=rocket 'CFLAGS_FOR_TARGET=-Os  -mcmodel=medany' 'CXXFLAGS_FOR_TARGET=-Os  -mcmodel=medan
y'                                                                                                                             
Thread model: single                                                                                                           
gcc version 9.2.0 (GCC) 
```

BernardXiong, 2022-11-18:
可以这样设置
```
set RTT_EXEC_PATH=D:/100-software/NucleiStudio/toolchain/gcc/bin
```

并确保下这个工具链存在。另外也可以编辑rtconfig.py中的配置，注意prefix是否正确等。

另外，如果你的环境是WSL，建议你使用纯粹的Linux环境或直接的Windows环境。因为WSL/WSL2问题太多，我们都不会在上面使用并验证。

closed 2022-11-27 by openluopworld