RT-Thread/rt-thread #6612: Flash接口R_FLASH_LP_AccessWindowSet使用不当导致芯片锁死

LinChenji, 2022-11-12:
代码如下：
`  {      
          err = flash_lp_init ();
          uint32_t wrWord=0x11223344;
          err = R_FLASH_LP_Erase(&g_flash_ctrl, FLASH_CF_BLOCK(8), 1);
          if(true == g_b_flash_event_erase_complete)
          {
                   R_BSP_SoftwareDelay(50, BSP_DELAY_UNITS_MILLISECONDS);
                    //err = R_FLASH_LP_AccessWindowSet(&g_flash_ctrl, FLASH_CF_START_ADDR(8), FLASH_CF_END_ADDR(8));//code:0x4000
                   err = R_FLASH_LP_Write(&g_flash_ctrl, (uint32_t )&wrWord, FLASH_CF_BLOCK(8), 1);
          }
          APP_PRINT("\r\n ** Performing Code Flash operations **\r\n");

          /* Close Flash_LP driver */
          err = flash_lp_deinit ();
      while (true)
      {
          APP_PRINT("\r\nR7FA2L1:Time 500ms elapse");
          R_PORT5->PODR ^= 1<<(BSP_IO_PORT_05_PIN_02 & 0xFF);
          R_PORT5->PODR ^= 1<<(BSP_IO_PORT_05_PIN_01 & 0xFF);
          R_BSP_SoftwareDelay(500, BSP_DELAY_UNITS_MILLISECONDS); //延时500毫秒
      }
}`
之后就一直JLINK擦除不掉

mysterywolf, 2022-11-12:
这是哪个bsp啊？ 哪个文件的代码？可否发个链接？

LinChenji, 2022-11-14:
Dear Meco:
你好，由于针对RA2L1的BSP最新止到3.6.0，PCB编号是CPK_RA2L1，而我下下来的例程是针对EK_RA2L1，BSP是4.0.0，我是基于EK_RA2L1例程来改的，所以会找不到版型，但我没有使用BSP耦合的数据结构和接口就没有报错，然后模仿例程写入Flash连续64字节，操作成功了，然后我改成写入一个字（4B），写法在issue里，然后就PC跑飞指向0xEFFFFFFE，再调试无法擦除和写入，Flash不知为何锁死了，几个状态寄存器见附件

mysterywolf, 2023-01-29:
这个和操作系统没什么关系吧，看你应该是全部调用的HAL库的api，在正常裸机可以擦除掉？

closed 2023-02-11 by mysterywolf