RT-Thread/rt-thread #6668: gcc10.3.1 报提醒  warning: 'strncpy' specified bound depends on the length of the source argument

yuqingli05, 2022-11-29:
版本 SHA-1: f57eb4feb5ee4856299058058eff15593ac0570d
gcc10.3.1 报提醒 
~~~
#define HAVE_NEWLIB_H 1
#define LIBC_VERSION "newlib 4.1.0"

#define HAVE_SYS_SIGNAL_H 1
#define HAVE_SYS_SELECT_H 1
#define HAVE_PTHREAD_H 1

#define HAVE_FDSET 1
#define HAVE_SIGACTION 1
#define HAVE_SIGEVENT 1
#define HAVE_SIGINFO 1
#define HAVE_SIGVAL 1
#define GCC_VERSION_STR "10.3.1 20210824 (release)"
#define STDC "2011"
~~~
~~~
F:\folder\rtt\rt-thread\include/rtthread.h:634:37: warning: 'strncpy' specified bound depends on the length of the source argument [-Wstringop-overflow=]
  634 | #define rt_strncpy(dest, src, n)    strncpy(dest, src, n)
      |                                     ^~~~~~~~~~~~~~~~~~~~~
F:\folder\rtt\rt-thread\components\drivers\spi\spi_flash_sfud.c:326:9: note: in expansion of macro 'rt_strncpy'
  326 |         rt_strncpy(spi_flash_dev_name_bak, spi_flash_dev_name, rt_strlen(spi_flash_dev_name));
      |         ^~~~~~~~~~
F:\folder\rtt\rt-thread\include/rtthread.h:634:37: warning: 'CC build\kernel\components\libc\compilers\common\cctype.o
strncpy' specified bound depends on the length of the source argument [-Wstringop-overflow=]
  634 | #define rt_strncpy(dest, src, n)    strncpy(dest, src, n)
      |                                     ^~~~~~~~~~~~~~~~~~~~~
F:\folder\rtt\rt-thread\components\drivers\spi\spi_flash_sfud.c:327:9: note: in expansion of macro 'rt_strncpy'
  327 |         rt_strncpy(spi_dev_name_bak, spi_dev_name, rt_strlen(spi_dev_name));
      |         ^~~~~~~~~~
~~~

dongly, 2022-12-01:
哪个bsp? 应是这个原因: #6665

yuqingli05, 2022-12-05:
> 哪个bsp? 应是这个原因: #6665

并不是特定的bsp，这个问题存在于 sfud 代码里面。strncpy，第三个参数是目的字符串的最大长度，用来防止写入溢出的问题。上面用法 strlen是计算的第二个源字符串长度，这种用法违反了 strncpy的本质目的，让strncpy退化为strcpy。建议使用strcpy， 或者strncpy第三个参数 用来指示 目的字符串的最大长度，而不是源字符串的长度。