RT-Thread/rt-thread #6680: NUC980 编译错误，BSP下未修改直接按步骤操作

kylongmu, 2022-12-01:
> menuconfig --generate

> pkgs --update
[Use Gitee server]
Cloning into 'D:\Develop\Git_Project\rt-thread\bsp\nuvoton\nk-980iot\packages\wavplayer-latest'...
remote: Enumerating objects: 98, done.
remote: Counting objects: 100% (40/40), done.
remote: Compressing objects: 100% (40/40), done.
remote: Total 98 (delta 14), reused 0 (delta 0), pack-reused 58
Unpacking objects: 100% (98/98), 1.64 MiB | 1.44 MiB/s, done.
==============================>  WAVPLAYER latest is downloaded successfully.

Cloning into 'D:\Develop\Git_Project\rt-thread\bsp\nuvoton\nk-980iot\packages\uffs-latest'...
remote: Enumerating objects: 105, done.
remote: Total 105 (delta 0), reused 0 (delta 0), pack-reused 105Receiving objects:  91% (96/105), 716.01 KiB | 219.00
Receiving objects: 100% (105/105), 854.16 KiB | 228.00 KiB/s, done.
Resolving deltas: 100% (22/22), done.
==============================>  UFFS latest is downloaded successfully.

Cloning into 'D:\Develop\Git_Project\rt-thread\bsp\nuvoton\nk-980iot\packages\ramdisk-latest'...
remote: Enumerating objects: 40, done.
remote: Counting objects: 100% (3/3), done.
remote: Compressing objects: 100% (3/3), done.
remote: Total 40 (delta 0), reused 0 (delta 0), pack-reused 37
Unpacking objects: 100% (40/40), 13.28 KiB | 56.00 KiB/s, done.
==============================>  RAMDISK latest is downloaded successfully.

Cloning into 'D:\Develop\Git_Project\rt-thread\bsp\nuvoton\nk-980iot\packages\optparse-latest'...
remote: Enumerating objects: 36, done.
remote: Counting objects: 100% (11/11), done.
remote: Compressing objects: 100% (11/11), done.

Unpacking objects: 100% (36/36), 14.20 KiB | 137.00 KiB/s, done.
==============================>  OPTPARSE latest is downloaded successfully.

==============================>  wavplayer update done

==============================>  uffs update done

==============================>  ramdisk update done

==============================>  optparse update done

Operation completed successfully.

> scons

In file included from d:\develop\rt_thread\env-windows\tools\gnu_gcc\arm_gcc\mingw\arm-none-eabi\include\sys\_default_fcntl.h:211,
                 from D:\Develop\Git_Project\rt-thread\components\libc\compilers\newlib/fcntl.h:14,
                 from D:\Develop\Git_Project\rt-thread\components\dfs\include/dfs.h:19,
                 from D:\Develop\Git_Project\rt-thread\components\dfs\include/dfs_fs.h:14,
                 from packages\uffs-latest\dfs_uffs.c:16:
packages\uffs-latest\src\inc/uffs/uffs_fd.h:100:18: error: expected ':', ',', ';', '}' or '__attribute__' before '.' token
  100 |     unsigned int st_atime;   /* time of last access */
      |                  ^~~~~~~~
packages\uffs-latest\dfs_uffs.c: In function 'dfs_uffs_stat':
packages\uffs-latest\dfs_uffs.c:607:21: error: 'struct uffs_stat' has no member named 'st_mtim'
  607 |     st->st_mtime = s.st_mtime;
      |                     ^
scons: *** [build\packages\uffs-latest\dfs_uffs.o] Error 1
scons: building terminated because of errors.

kylongmu, 2022-12-02:
能加一个nuvoton bsp的action吗？

wosayttn, 2022-12-29:
Fixed in #6793. FYI. Thanks.

closed 2022-12-29 by mysterywolf