RT-Thread/rt-thread #6726: 控制libcpu下的risc-v代码

BernardXiong, 2022-12-11:
目前libcpu/risc-v下的代码，基本上接入一家芯片厂商或芯片就会来一份文件夹，后面需要严格控制者部分的代码，尽可能合并。如果需要独立的文件夹，需要给出完善，完整的理由出来，否则不会进行合并。

同时开启libcpu/risc-v下代码的合并，可以是这样的模式：
```text
libcpu/risc-v
  \- common
  |
  \- xx_vendor
      \- common
      |
     \- chip
```

尽可能的重用代码，例如`rt_hw_stack_init`, `rt_hw_interrupt_disable/enable`等

BernardXiong, 2022-12-11:
以下这些代码都是需要整理的，这些[代码](https://github.com/RT-Thread/rt-thread/blob/master/libcpu/risc-v/SConscript)都跳过了common目录下的代码

``` python
# add common code files
if   rtconfig.CPU == "e9xx" :
    group = group
elif rtconfig.CPU == "nuclei" :
    group = group
elif rtconfig.CPU == "virt64" :
    group = group
elif rtconfig.CPU == "c906" :
    group = group
elif rtconfig.CPU == "ch32v1" :
    group = group
elif rtconfig.CPU == "ch32v3" :
    group = group
elif rtconfig.CPU == "hpmicro":
    group = group
elif rtconfig.CPU == "cv32e40p":
    group = group
else :
    group = group + SConscript(os.path.join('common', 'SConscript'))
```

polarvid, 2022-12-12:
是否可以用 cpu ISA 的形式组织。riscv ISA 的常用的几个标准比如 2.2, 20190608, 20191213... 可以具体看一下相关的兼容性，然后以文件夹的形式组织起来。比如 qemu-virt64 就可以包括 `rv64ima`, `rv64v_1_0` 这些文件夹。c908 就可以包括 `rv64ima`, `rv64v_1_0`, [`xthead_*`](https://[github.com](https://github.com/T-head-Semi/thead-extension-spec)/T-head-Semi/thead-extension-spec)

BernardXiong, 2023-03-11:
这个issue是否可以关闭了？

closed 2023-03-11 by mysterywolf