RT-Thread/rt-thread #6730: FAL在开启 分区表是否在 fal_cfg.h 中定义时未将分区表主动保存至flash

i-jaffer, 2022-12-12:
在使用FAL时，开启 `分区表在 fal_cfg.h 中定义`，目前程序未将分区表信息主动写入flash，那么如果像bootload也需要使用分区表，目前方案只能拷贝app的  `fal_cfg.h` 文件

是否考虑在 开启  `分区表在 fal_cfg.h 中定义` 时，主动将分区表写入flash，之后如果bootload不开启  `分区表在 fal_cfg.h 中定义`， 则可以直接从flash中加载

Guozhanxin, 2022-12-22:
fal本身是有自动在flash上检索分区表的功能的。
[image]

i-jaffer, 2022-12-24:
> fal本身是有自动在flash上检索分区表的功能的。 [image]

目前分区表信息没有保存在flash中，比如在menuconfig中开启`分区表在 fal_cfg.h 中定义`，分区表我看程序里面是存在ram里面的，没有将分区信息写到flash，那么如果menuconfig中关闭了`分区表在 fal_cfg.h 中定义`，程序会自动去flash里面找分区表，实际会是找不到的，是否考虑把分区表信息往flash中写一份呢？