RT-Thread/rt-thread #6761: 多媒体package->mp3player编译报错

imntjempty, 2022-12-20:
[package->mp3player]
rt-thread版本 4.0.5
bsp 某个ARM BSP
工具链 arm-none-eabi-gcc -v(5.4.1) 
复现步骤：当在scons中的menuconfig选中mp3player时（PKG_USING_MP3PLAYER），编译报错

```
CC build\packages\wavplayer-latest\src\wavrecorder.o
CC build\packages\wavplayer-latest\src\wavrecorder_cmd.o
LINK rt-thread.elf
d:/env/tools/gnu_gcc/arm_gcc/mingw/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/bin/ld.exe: error: build\packages\helix-v1.0.0\real\arm\asmmisc_gcc.o: Conflicting architecture profiles M/A
d:/env/tools/gnu_gcc/arm_gcc/mingw/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/bin/ld.exe: failed to merge target specific data of file build\packages\helix-v1.0.0\real\arm\asmmisc_gcc.o
d:/env/tools/gnu_gcc/arm_gcc/mingw/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/bin/ld.exe: error: build\packages\helix-v1.0.0\real\arm\asmpoly_thumb2_gcc.o: Conflicting architecture profiles M/A
d:/env/tools/gnu_gcc/arm_gcc/mingw/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/bin/ld.exe: failed to merge target specific data of file build\packages\helix-v1.0.0\real\arm\asmpoly_thumb2_gcc.o
collect2.exe: error: ld returned 1 exit status
scons: *** [rt-thread.elf] Error 1
scons: building terminated because of errors.
```

BernardXiong, 2022-12-21:
`bsp 某个ARM BSP`

这个需要给出更多些信息，ARM Cortex-M？或者ARM Cortex-A （32），ARM Cortex-A（64）

imntjempty, 2022-12-21:
> `bsp 某个ARM BSP`
> 
> 这个需要给出更多些信息，ARM Cortex-M？或者ARM Cortex-A （32），ARM Cortex-A（64）

cortex-a7

mysterywolf, 2023-02-09:
请问是哪个BSP 为什么是某个BSP？

closed 2023-03-14 by mysterywolf