RT-Thread/rt-thread #6768: 在shell中使用ps命令查询线程状态时，由于rt_thread_sleep中临时改变状态会导致ps出的状态与实际状态不一致，影响判断

Yaochenger, 2022-12-22:
**现象**

[image]
**出现问题的地方**

[image]

**描述：上述临时改变错误码的操作会导致ps的时候显示出这个错误码，影响用户判断线程状态。**

wdfk-prog, 2022-12-22:
- 我也是有这样情况
[image]
- 另一个你的error和ok的颜色是怎么改出来的?

mysterywolf, 2022-12-22:
这块我会重新处理一下posix和rtt error code的转换

heyuanjie87, 2022-12-23:
我也正纳闷好端端的线程咋现在出来那么多eintr的，是该处理下了

zhkag, 2022-12-23:
这是信号打断延时的时候改的，可能没有考虑完全，改出问题了

a1012112796, 2023-02-08:
确实不是很明白此处设`thread->error = -RT_EINTR;` 有何作用?

Guozhanxin, 2023-03-25:
https://github.com/RT-Thread/rt-thread/pull/7096 大家看看这个pr

logeexpluoqi, 2023-11-14:
这个问题在5.0.2中还存在

/>$ ps
object_loc thread   pri  status      sp     stack size max used left tick  error
---------- -------- ---  ------- ---------- ----------  ------  ---------- ---
0x20003188 injector   7  suspend 0x00000068 0x00000400    10%   0x00000005 EINTRPT
0x20002eac link      10  suspend 0x00000074 0x00000200    28%   0x00000003 EINTRPT
0x20002a08 uart3_rx   6  suspend 0x000000b0 0x00000400    17%   0x00000005 EINTRPT
0x20002740 blink      5  suspend 0x00000068 0x00000100    40%   0x00000001 EINTRPT
0x20001e04 msh       20  running 0x00000204 0x00000800    32%   0x00000006 OK
0x20000a04 tidle0    31  ready   0x00000060 0x00000100    40%   0x0000001a OK
0x20000e28 timer      4  suspend 0x00000054 0x00000200    16%   0x00000009 OK
/>$