RT-Thread/rt-thread #6777: 编译报错  F_OK找不到

yuqingli05, 2022-12-25:
SHA-1: fa7c1db76874dfb97bc3673826d716aa29eefa6b
测试BSP: qemu-vexpress-a9 （其实和BSP无关 具有普遍性）
编译器 arm-none-eabi-gcc
~~~c
#include <stdint.h>
#include <stdio.h>
#include <stdlib.h>
#include <unistd.h>

int main(void)
{
    if (access("/test/test", F_OK) == 0)
    {
    }
    return 0;
}
~~~
[image]

原因分析：
rtthread 头文件unistd.h 里面没有包含 F_OK定义
删除 rt-thread\components\libc\compilers\common\include\sys\unistd.h 和 rt-thread\components\libc\compilers\common\include\unistd.h 后正常。
删除rtthread 的 unistd.h 文件后会使用 工具链的 unistd.h 文件，编译通过
在gcc工具连上 还有必要用rtthread自带的unistd.h吗？

mysterywolf, 2022-12-25:
感谢反馈，会增加上

mysterywolf, 2022-12-26:
https://github.com/RT-Thread/rt-thread/pull/6778

yuqingli05, 2022-12-26:
> #6778

我将 rtthread 目录下的 unistd,h 删除之后，也能编译成功。原因是 gcc 头文件目录下也有这个头文件。rtthread 提供的和gcc 自身的有什么区别那？

mysterywolf, 2022-12-26:
rt-thread的unistd.h是通用的，服务于多个不同的工具链。因为不同工具链有的提供了 有的没提供，提供的程度也参差不齐，所以接管到了rt-thread的unistd.h中

closed 2023-01-02 by mysterywolf