RT-Thread/rt-thread #6814: at框架调用at_client_recv会引起串口接收数据冲突

i-jaffer, 2023-01-04:
在 `at_client_para_init` 初始化之后会创建`client_parser`线程，负责监听 rx 的数据
client_parser
&emsp; -> at_recv_readline
&emsp;&emsp; -> at_client_getchar(client, &ch, RT_WAITING_FOREVER);
&emsp;&emsp;&emsp; -> rt_device_read / rt_sem_take(client->rx_notice,..)

同时at组件对外开放以下接口：
```c
#define at_exec_cmd(resp, ...)                   at_obj_exec_cmd(at_client_get_first(), resp, __VA_ARGS__)
#define at_client_wait_connect(timeout)          at_client_obj_wait_connect(at_client_get_first(), timeout)
#define at_client_send(buf, size)                at_client_obj_send(at_client_get_first(), buf, size)
#define at_client_recv(buf, size, timeout)       at_client_obj_recv(at_client_get_first(), buf, size, timeout)
#define at_set_end_sign(ch)                      at_obj_set_end_sign(at_client_get_first(), ch)
#define at_set_urc_table(urc_table, table_sz)    at_obj_set_urc_table(at_client_get_first(), urc_table, table_sz)
```
如果调用 `at_client_recv` ，则也会在此函数内调用`rt_sem_take(client->rx_notice,..)`以及`rt_device_read`

此会导致两个线程争抢`client->rx_notice`信号量和对应串口 rx 的数据

i-jaffer, 2023-01-07:
[image]
[image]
以上两处位置存在冲突

JasonCang, 2023-06-02:
不冲突，obj_recv只能在urc回调中使用