RT-Thread/rt-thread #6815: 新唐nk-rtu980用iperf测试以太网，nk-rtu980作为服务端速率只能测到几百K

lfkabc, 2023-01-05:
详细截图与log见https://club.rt-thread.org/ask/question/be3978bc89c31b70.html。

BernardXiong, 2023-01-06:
这个得找新唐官方支持才行，这个bsp是由新唐工程师维护的。

wosayttn, 2023-01-07:
Hi,

The EMAC driver implemented zero-copy and NAPI-like method to improve the RX throughput.
Maybe you have a testing using lwiperf on RTU980.

See the description to bring up lwiperf utility. Thanks.
https://github.com/RT-Thread/rt-thread/blob/master/bsp/nuvoton/libraries/nuc980/rtt_port/drv_emac.c#L667

tomyqg, 2023-01-09:
[
[image]
](url)

wosayttn, 2023-01-09:
Thanks Tomyqg.

Try to adjust optimization level to O2 by disabling DEBUG in rtconfig.py.
See here, https://github.com/RT-Thread/rt-thread/blob/master/bsp/nuvoton/nk-rtu980/rtconfig.py#L27 

Also see my testing after disabling debug mode.
[image]

tomyqg, 2023-01-09:
用lwiperf可以测到一样的带宽，但是rtt自带的iperf(基于socket api)测速如何提高？

tomyqg, 2023-01-10:
[drv_emac.txt](https://github.com/RT-Thread/rt-thread/files/10378370/drv_emac.txt)
这些优化效果甚微，最新这个emac驱动用socket [api测速还是很慢，但是之前这个驱动 开O2优化，rtt iperf 测速还可以[](url)]([image])

tomyqg, 2023-01-10:
最新的emac驱动，通过调整lwip配置，主要是下面几项，iper测速带宽有所提升,但是和老版驱动还是有较大差距
 #define RT_LWIP_TCP_SEG_NUM 256
#define RT_LWIP_TCP_SND_BUF 32768
#define RT_LWIP_TCP_WND 10240

[image]

wosayttn, 2023-01-10:
Hi, @tomyqg 

Latest version applied zero-copy RX in EMAC driver. It is better that previous memcpy version for saving bandwidth.
We also updated new default setting for the iperf issue. The score is up to 60Mbps in my testing.

lfkabc, 2023-01-15:
最新代码OK。

closed 2023-03-23 by Guozhanxin