RT-Thread/rt-thread #6817: [nuvoton arduino]编译出错，应该是原厂还没支持

kylongmu, 2023-01-05:
拉取最新rt-thread，menuconfig里选“adafruit pcf8574”，然后编译，测试了m467,m487,nuk980几个bsp都是一样的。
../packages/perf_counter-latest/perf_counter.c:24:28: fatal error: cmsis_compiler.h: No such file or directory
compilation terminated.
arm-none-eabi-g++ "../packages/RTduino-latest/core/Stream.cpp"
In file included from ../packages/perf_counter-latest/os/perf_os_patch_rt_thread.c:23:0:
D:\RT-ThreadStudio\workspace\numaker-pfm-m487-50\packages\perf_counter-latest/perf_counter.h:25:28: fatal error: cmsis_compiler.h: No such file or directory
compilation terminated.
arm-none-eabi-g++ "../packages/RTduino-latest/core/Tone.cpp"
make: *** [packages/perf_counter-latest/subdir.mk:18: packages/perf_counter-latest/perf_counter.o] Error 1
make: *** Waiting for unfinished jobs....
make: *** [packages/perf_counter-latest/os/subdir.mk:18: packages/perf_counter-latest/os/perf_os_patch_rt_thread.o] Error 1
In file included from ../packages/RTduino-latest/libraries/buildin/Wire/Wire.h:39:0,
                 from ../packages/RTduino-latest/libraries/buildin/Wire/Wire.cpp:40:
D:\RT-ThreadStudio\workspace\numaker-pfm-m487-50\packages\RTduino-latest\core/Arduino.h:48:26: fatal error: pins_arduino.h: No such file or directory
compilation terminated.
make: *** [packages/RTduino-latest/libraries/buildin/Wire/subdir.mk:18: packages/RTduino-latest/libraries/buildin/Wire/Wire.o] Error 1
In file included from ../packages/RTduino-latest/core/Print.cpp:33:0:
../packages/RTduino-latest/core/Arduino.h:48:26: fatal error: pins_arduino.h: No such file or directory
compilation terminated.
make: *** [packages/RTduino-latest/core/subdir.mk:61: packages/RTduino-latest/core/Print.o] Error 1
In file included from ../packages/RTduino-latest/core/RTduino.c:13:0:
D:\RT-ThreadStudio\workspace\numaker-pfm-m487-50\packages\RTduino-latest\core/Arduino.h:48:26: fatal error: pins_arduino.h: No such file or directory
compilation terminated.
In file included from ../packages/RTduino-latest/core/HardwareSerial.cpp:12:0:
../packages/RTduino-latest/core/HardwareSerial.h:18:26: fatal error: pins_arduino.h: No such file or directory
compilation terminated.
make: *** [packages/RTduino-latest/core/subdir.mk:63: packages/RTduino-latest/core/RTduino.o] Error 1
make: *** [packages/RTduino-latest/core/subdir.mk:61: packages/RTduino-latest/core/HardwareSerial.o] Error 1
In file included from ../packages/RTduino-latest/core/Stream.cpp:34:0:
../packages/RTduino-latest/core/Arduino.h:48:26: fatal error: pins_arduino.h: No such file or directory
compilation terminated.
make: *** [packages/RTduino-latest/core/subdir.mk:61: packages/RTduino-latest/core/Stream.o] Error 1
In file included from ../packages/RTduino-latest/core/wiring_private.h:14:0,
                 from ../packages/RTduino-latest/core/Tone.cpp:12:
D:\RT-ThreadStudio\workspace\numaker-pfm-m487-50\packages\RTduino-latest\core/Arduino.h:48:26: fatal error: pins_arduino.h: No such file or directory
compilation terminated.
make: *** [packages/RTduino-latest/core/subdir.mk:61: packages/RTduino-latest/core/Tone.o] Error 1
"make -j12 all" terminated with exit code 2. Build might be incomplete.

16:27:48 Build Failed. 17 errors, 0 warnings. (took 29s.552ms)

mysterywolf, 2023-01-05:
该bsp还没有完整支持Arduino pinout，这个是已经支持了的bsp
https://github.com/RTduino/RTduino/blob/master/README_zh.md#11-%E5%B7%B2%E7%BB%8F%E6%94%AF%E6%8C%81arduino%E7%94%9F%E6%80%81%E5%85%BC%E5%AE%B9%E5%B1%82%E7%9A%84rt-thread-bsp

如果是没有支持的bsp，还想用IIC类的Arduino库，可以开启精简模式：
https://github.com/RTduino/RTduino/blob/master/README_zh.md#5-rtduino%E7%B2%BE%E7%AE%80%E6%A8%A1%E5%BC%8F%E5%BF%AB%E9%80%9F%E4%BD%BF%E7%94%A8%E6%97%A0%E9%9C%80%E9%80%82%E9%85%8D%E7%89%B9%E5%AE%9Absp

kylongmu, 2023-01-06:
> 该bsp还没有完整支持Arduino pinout，这个是已经支持了的bsp https://github.com/RTduino/RTduino/blob/master/README_zh.md#11-%E5%B7%B2%E7%BB%8F%E6%94%AF%E6%8C%81arduino%E7%94%9F%E6%80%81%E5%85%BC%E5%AE%B9%E5%B1%82%E7%9A%84rt-thread-bsp
> 
> 如果是没有支持的bsp，还想用IIC类的Arduino库，可以开启精简模式： https://github.com/RTduino/RTduino/blob/master/README_zh.md#52-%E5%A6%82%E4%BD%95%E4%B8%8D%E7%94%A8%E5%AE%9A%E4%B9%89%E5%BC%95%E8%84%9A%E6%98%A0%E5%B0%84%E8%A1%A8%E6%9B%B4%E6%96%B9%E4%BE%BF%E7%9A%84%E4%BD%BF%E7%94%A8rtduino
> 
> 有问题可以继续提问

多谢帮助，拉取最新rt-thread，numaker-pfm-m487开发板，打开adafruit pcf8574，按照精简模式使用使能以下[Don't use setup-loop] [Enable tiny mode]后编译出错：
linking...
./rt-thread/components/libc/cplusplus/cxx_crt_init.o: In function `cplusplus_system_init':
cxx_crt_init.c:(.text.cplusplus_system_init+0x18): undefined reference to `__ctors_start__'
cxx_crt_init.c:(.text.cplusplus_system_init+0x1c): undefined reference to `__ctors_end__'
collect2.exe: error: ld returned 1 exit status
make: *** [makefile:116: rtthread.elf] Error 1
"make -j12 all" terminated with exit code 2. Build might be incomplete.
这个错误看起来是rt-thread主线内的。

BernardXiong, 2023-01-06:
`ctors_start/ctors_end`应该是C++中和链接脚本相关的，可以看看gcc链接脚本的情况。

mysterywolf, 2023-01-06:
说明这个bsp没有支持基本的C++执行环境

kylongmu, 2023-01-06:
> 说明这个bsp没有支持基本的C++执行环境

看来对我超纲了，还是等厂家支持吧。