RT-Thread/rt-thread #6874: mmc/sdio块设备移除分区偏移的问题

BernardXiong, 2023-01-23:
在合并smart代码的过程中，mmc/sdio块设备把分区偏移移除，交给上层文件系统来处理。

这样导致：
* 上层可以通过偏移的方式访问不属于这个块设备的其他区域；
* 上层进行块设备格式化时，对分区表处理迷乱了；
* 涉及文件系统：elm fatfs，ext4等文件系统；
* 并且，所有涉及到块设备 + 分区表的地方都错乱了。

BernardXiong, 2023-01-23:
elm fatfs移植的disk层未对分区进行处理，所以目前elm fatfs版本不支持多分区。

mysterywolf, 2023-01-23:
社区小伙伴也报告过这个问题，之前smart合并之前用的没问题，但是smart合并之后，sd这块的代码逻辑貌似发生了变化。

BernardXiong, 2023-01-23:
把mmc/sd块设备的分区偏移加回来，fatfs多分区的问题可以解决。还需要修正lwext4的问题

BernardXiong, 2023-01-23:
lwext4的修正： https://github.com/RT-Thread-packages/lwext4/commit/dee7ee5e7a604a921425d01abbf6102fcd8450fd

BernardXiong, 2023-01-25:
在qemu上完成了single ext, single fat, dual ext, ext + fat, fat + ext, fat + fat的测试验证，后面 @Guozhanxin 也可以帮忙做个真机验证，都确认OK后，就可以把这个issue关闭了。

mysterywolf, 2023-01-30:
https://github.com/RT-Thread/rt-thread/pull/6886
https://github.com/geniusgogo/littlefs/pull/12
https://github.com/geniusgogo/littlefs/pull/13
郭老师 这几个PR也一起看一下吧 文件系统改了写什么我最近没有跟踪 @Guozhanxin

Guozhanxin, 2023-02-08:
> 在qemu上完成了single ext, single fat, dual ext, ext + fat, fat + ext, fat + fat的测试验证，后面 @Guozhanxin 也可以帮忙做个真机验证，都确认OK后，就可以把这个issue关闭了。

这个应该怎么验证呢？分区表指的是什么？gpt分区吗？

closed 2024-04-20 by BernardXiong