RT-Thread/rt-thread #6881: Convert superior sine wave to square using 2 channel oscilloscope - CH32V307 Project issue

neuberfran, 2023-01-28:
Hi,

[image]

I've been trying to generate this wave for 2 weeks in my hanmatek dos1102.

**The whole project is based on the post below, by Mr Lars:**
https://stm32world.com/wiki/CH32_Oscillator#Oscillator_Calculation

**Yesterday I got the Oscillator project (note that I don't use 1V or 1,000.00Hz)**
https://www.youtube.com/watch?v=re7jdh8S404
[image]

**Now, I need help changing my current project:**
https://github.com/neuberfran/CH32V307V-EVT-R1/tree/master/oscillator
**To get the waveforms from the figure at the top of this issue, using 1V and 1,000.00Hz**

Tks

mysterywolf, 2023-01-28:
It's amazing, but I don't get that is there any problem when you are using RT-Thread?

neuberfran, 2023-01-29:
@mysterywolf 

Come on, there are several things to be said but first I
 would say that to upload the Oscillator project (original by Mr. Lars)
 I had to use MounRiver Studio_Community and
MounRiver Studio, in the latter: right click 
on the oscillator.wvproj file -> open with MounRiver Studio.

First of all, a question: What does a project necessarily need 
to have to be an RT-Thread project?

The project below:
https://github.com/neuberfran/test2
has the rtthread folder. And I created it to port the original Oscillator 
from Mr. Lars. Of course, it replaces several files in the User folder,
etc... The way I got it was this one below:
https://www.youtube.com/watch?v=2qNC_agaofM

**And it's not:**
https://www.youtube.com/watch?v=re7jdh8S404
**which would be correct**

As you can see in the picture below, I created it using MounRiver 
and RT-Thread OS option:
[image]

You can take this project (test2 project) and have it produce the correct waveforms.
It would be a way and lastly to make it produce the waveforms that I want 
quoted at the beginning of this issue

I could also switch to the RT-Thread Studio IDE. Could you give me the steps 
for me (using Mr Lars' project, or not) to produce the waveforms from the
 beginning of this issue.

mysterywolf, 2023-01-29:
If my understanding is correct, it's a specific issue of "MounRiver Studio Community", right? You may ask this community....

neuberfran, 2023-01-29:
@mysterywolf 
Can you make my test2 project produce oscilloscope waves the same as Mr. Lars?
https://github.com/neuberfran/test2

https://www.youtube.com/watch?v=re7jdh8S404

mysterywolf, 2023-01-29:
Sorry, I cannot. Because generate waves is a specific problem or question related to the hardware itself instead of RT-Thread OS, and I don't have the hardware board in my hand.

mysterywolf, 2023-01-29:
I think this issue open in a wrong place. Issue is used for reporting some bugs when you are using RT-Thread itself, rather than as long as you used rt-thread and no matter the problem is related rt-thread or not in actual, you always report in our community. In github issue, we only handle the problems/bugs related rt-thread in actual.

neuberfran, 2023-01-29:
@mysterywolf 

You can now, as a demo, create a project from scratch using the RT-Thread Studio ide and the ch32v307 device to generate the waveform below:

[image]
Yes or no?

neuberfran, 2023-02-05:
Tks Sir  @qiyongshuang

https://github.com/qiyongshuang/test2-main

https://github.com/neuberfran/test2

closed 2023-02-05 by neuberfran