RT-Thread/rt-thread #6882: 拉取最新代码后bsp下的qemu-virt64-aarch64编译失败

huqiang897, 2023-01-29:
commit: b3105414717a6670b770621f2d0cd2d6eee18e02
编译环境：Ubuntu22.04
拉了最新的master代码后，直接scons编译报错，如下：
scons: Reading SConscript files ...
scons: done reading SConscript files.
scons: Building targets ...
scons: building associated VariantDir targets: build
CC build/applications/main.o
In file included from /home/qiang/Temp/rt-thread/components/libc/compilers/common/include/sys/time.h:18,
                 from /home/qiang/Temp/rt-thread/components/libc/compilers/common/include/sys/select.h:17,
                 from /opt/toolchain/gcc-arm-10.2-2020.11-x86_64-aarch64-none-elf/aarch64-none-elf/include/sys/types.h:50,
                 from /opt/toolchain/gcc-arm-10.2-2020.11-x86_64-aarch64-none-elf/aarch64-none-elf/include/stdio.h:61,
                 from applications/main.c:11:
/opt/toolchain/gcc-arm-10.2-2020.11-x86_64-aarch64-none-elf/aarch64-none-elf/include/time.h:56:1: error: unknown type name 'clock_t'; did you mean '_flock_t'?
   56 | clock_t    clock (void);
      | ^~~~~~~
      | _flock_t
In file included from /home/qiang/Temp/rt-thread/components/libc/compilers/common/include/sys/select.h:17,
                 from /opt/toolchain/gcc-arm-10.2-2020.11-x86_64-aarch64-none-elf/aarch64-none-elf/include/sys/types.h:50,
                 from /opt/toolchain/gcc-arm-10.2-2020.11-x86_64-aarch64-none-elf/aarch64-none-elf/include/stdio.h:61,
                 from applications/main.c:11:
/home/qiang/Temp/rt-thread/components/libc/compilers/common/include/sys/time.h:56:5: error: unknown type name 'suseconds_t'
   56 |     suseconds_t tv_usec;    /* and microseconds */
      |     ^~~~~~~~~~~
/home/qiang/Temp/rt-thread/components/libc/compilers/common/include/sys/time.h:141:20: error: unknown type name 'clockid_t'
  141 | int clock_getres  (clockid_t clockid, struct timespec *res);
      |                    ^~~~~~~~~
/home/qiang/Temp/rt-thread/components/libc/compilers/common/include/sys/time.h:142:20: error: unknown type name 'clockid_t'
  142 | int clock_gettime (clockid_t clockid, struct timespec *tp);
      |                    ^~~~~~~~~
/home/qiang/Temp/rt-thread/components/libc/compilers/common/include/sys/time.h:143:20: error: unknown type name 'clockid_t'
  143 | int clock_settime (clockid_t clockid, const struct timespec *tp);
      |                    ^~~~~~~~~
/home/qiang/Temp/rt-thread/components/libc/compilers/common/include/sys/time.h:144:21: error: unknown type name 'clockid_t'
  144 | int clock_nanosleep(clockid_t clockid, int flags, const struct timespec *rqtp, struct timespec *rmtp);
      |                     ^~~~~~~~~
scons: *** [build/applications/main.o] Error 1
scons: building terminated because of errors.

BernardXiong, 2023-01-29:
smart模式是否打开了？如果打开了，需要aarch64-musl-gcc的工具链。

huqiang897, 2023-01-29:
> smart模式是否打开了？如果打开了，需要aarch64-musl-gcc的工具链。

[image]
没设置的

BernardXiong, 2023-01-29:
```bash
CC build/kernel/src/clock.o
CC build/kernel/src/components.o
CC build/kernel/src/cpu.o
CC build/kernel/src/device.o
CC build/kernel/src/idle.o
CC build/kernel/src/ipc.o
CC build/kernel/src/irq.o
CC build/kernel/src/kservice.o
CC build/kernel/src/mem.o
CC build/kernel/src/memheap.o
CC build/kernel/src/mempool.o
CC build/kernel/src/object.o
CC build/kernel/src/scheduler.o
CC build/kernel/src/thread.o
CC build/kernel/src/timer.o
LINK rtthread.elf
aarch64-none-elf-objcopy -O binary rtthread.elf rtthread.bin
aarch64-none-elf-size rtthread.elf
   text	   data	    bss	    dec	    hex	filename
 297258	   3600	  90608	 391466	  5f92a	rtthread.elf
scons: done building targets.
~/workspace/rt-thread/bsp/qemu-virt64-aarch64$ aarch64-none-elf-gcc -v
Using built-in specs.
COLLECT_GCC=aarch64-none-elf-gcc
COLLECT_LTO_WRAPPER=/opt/gcc-arm-10.2-2020.11-x86_64-aarch64-none-elf/bin/../libexec/gcc/aarch64-none-elf/10.2.1/lto-wrapper
Target: aarch64-none-elf
Configured with: /tmp/dgboter/bbs/build02--cen7x86_64/buildbot/cen7x86_64--aarch64-none-elf/build/src/gcc/configure --target=aarch64-none-elf --prefix=/tmp/dgboter/bbs/build02--cen7x86_64/buildbot/cen7x86_64--aarch64-none-elf/build/build-aarch64-none-elf/install// --with-gmp=/tmp/dgboter/bbs/build02--cen7x86_64/buildbot/cen7x86_64--aarch64-none-elf/build/build-aarch64-none-elf/host-tools --with-mpfr=/tmp/dgboter/bbs/build02--cen7x86_64/buildbot/cen7x86_64--aarch64-none-elf/build/build-aarch64-none-elf/host-tools --with-mpc=/tmp/dgboter/bbs/build02--cen7x86_64/buildbot/cen7x86_64--aarch64-none-elf/build/build-aarch64-none-elf/host-tools --with-isl=/tmp/dgboter/bbs/build02--cen7x86_64/buildbot/cen7x86_64--aarch64-none-elf/build/build-aarch64-none-elf/host-tools --disable-shared --disable-nls --disable-threads --disable-tls --enable-checking=release --enable-languages=c,c++,fortran --with-newlib --with-pkgversion='GNU Toolchain for the A-profile Architecture 10.2-2020.11 (arm-10.16)' --with-bugurl=https://bugs.linaro.org/
Thread model: single
Supported LTO compression algorithms: zlib
gcc version 10.2.1 20201103 (GNU Toolchain for the A-profile Architecture 10.2-2020.11 (arm-10.16)) 
```

没问题呀，
```bash
:~/workspace/rt-thread/bsp/qemu-virt64-aarch64$ git status . 
位于分支 master
您的分支与上游分支 'origin/master' 一致。

无文件要提交，干净的工作区
```

closed 2023-01-30 by huqiang897

seanfight, 2023-09-18:
平台：mac m2
解释器：r'/Applications/ARM/bin/arm-none-eabi-gcc'
(base) donghuiyuan@donghuiyuandeMacBook-Pro stm32l475-atk-pandora % cd /Users/donghuiyuan/rt-thread-master/bsp/stm32/stm32l475-atk-pandora/dist/project
(base) donghuiyuan@donghuiyuandeMacBook-Pro project % scons
scons: Reading SConscript files ...
scons: done reading SConscript files.
scons: Building targets ...
scons: building associated VariantDir targets: build
CC build/applications/main.o
CC build/board/CubeMX_Config/Src/stm32l4xx_hal_msp.o
CC build/board/board.o
CC build/kernel/components/drivers/core/device.o
CC build/kernel/components/drivers/ipc/completion.o
CC build/kernel/components/drivers/ipc/dataqueue.o
CC build/kernel/components/drivers/ipc/pipe.o
CC build/kernel/components/drivers/ipc/ringblk_buf.o
CC build/kernel/components/drivers/ipc/ringbuffer.o
CC build/kernel/components/drivers/ipc/waitqueue.o
CC build/kernel/components/drivers/ipc/workqueue.o
CC build/kernel/components/drivers/misc/pin.o
CC build/kernel/components/drivers/serial/serial.o
CC build/kernel/components/finsh/cmd.o
CC build/kernel/components/finsh/msh.o
CC build/kernel/components/finsh/msh_parse.o
CC build/kernel/components/finsh/shell.o
In file included from rt-thread/components/libc/compilers/common/include/sys/time.h:18,
                 from rt-thread/components/libc/compilers/common/include/sys/select.h:17,
                 from /Applications/ARM/arm-none-eabi/include/sys/types.h:50,
                 from /Applications/ARM/arm-none-eabi/include/stdio.h:61,
                 from rt-thread/components/finsh/shell.c:23:
/Applications/ARM/arm-none-eabi/include/time.h:56:1: error: unknown type name 'clock_t'; did you mean '_flock_t'?
   56 | clock_t    clock (void);
      | ^~~~~~~
      | _flock_t
In file included from rt-thread/components/libc/compilers/common/include/sys/select.h:17,
                 from /Applications/ARM/arm-none-eabi/include/sys/types.h:50,
                 from /Applications/ARM/arm-none-eabi/include/stdio.h:61,
                 from rt-thread/components/finsh/shell.c:23:
rt-thread/components/libc/compilers/common/include/sys/time.h:81:5: error: unknown type name 'suseconds_t'
   81 |     suseconds_t tv_usec;    /* and microseconds */
      |     ^~~~~~~~~~~
scons: *** [build/kernel/components/finsh/shell.o] Error 1
scons: building terminated because of errors.
(base) donghuiyuan@donghuiyuandeMacBook-Pro project %

Akashsujanapal, 2023-10-14:
i have this same error

Akashsujanapal, 2023-10-14:
12:48:04 **** Build of configuration Debug for project projrct ****
make -j4 all2 
riscv64-unknown-elf-gcc "../rt-thread/src/clock.c"
riscv64-unknown-elf-gcc "../rt-thread/src/components.c"
riscv64-unknown-elf-gcc "../rt-thread/src/idle.c"
riscv64-unknown-elf-gcc "../rt-thread/src/ipc.c"
riscv64-unknown-elf-gcc "../rt-thread/src/irq.c"
riscv64-unknown-elf-gcc "../rt-thread/src/kservice.c"
riscv64-unknown-elf-gcc "../rt-thread/src/mem.c"
riscv64-unknown-elf-gcc "../rt-thread/src/memheap.c"
riscv64-unknown-elf-gcc "../rt-thread/src/mempool.c"
riscv64-unknown-elf-gcc "../rt-thread/src/object.c"
riscv64-unknown-elf-gcc "../rt-thread/src/scheduler_up.c"
riscv64-unknown-elf-gcc "../rt-thread/src/thread.c"
riscv64-unknown-elf-gcc "../rt-thread/src/timer.c"
riscv64-unknown-elf-gcc "../rt-thread/components/libc/compilers/common/cctype.c"
riscv64-unknown-elf-gcc "../rt-thread/components/libc/compilers/common/cstdlib.c"
riscv64-unknown-elf-gcc "../rt-thread/components/libc/compilers/common/cstring.c"
riscv64-unknown-elf-gcc "../rt-thread/components/libc/compilers/common/ctime.c"
riscv64-unknown-elf-gcc "../rt-thread/components/libc/compilers/common/cunistd.c"
riscv64-unknown-elf-gcc "../rt-thread/components/libc/compilers/common/cwchar.c"
In file included from D:\RT-ThreadStudio\workspace\projrct\rt-thread\components\libc\compilers\common\include/sys/select.h:17,
                 from d:\rt-threadstudio\repo\extract\toolchain_support_packages\risc-v\risc-v-gcc\11.1.0\riscv64-unknown-elf\include\sys\types.h:50,
                 from D:\RT-ThreadStudio\workspace\projrct\rt-thread\components\libc\compilers\common\include/sys/unistd.h:16,
                 from D:\RT-ThreadStudio\workspace\projrct\rt-thread\components\libc\compilers\common\include/unistd.h:14,
                 from ../rt-thread/components/libc/compilers/common/cunistd.c:12:
D:\RT-ThreadStudio\workspace\projrct\rt-thread\components\libc\compilers\common\include/sys/time.h:82:5: error: unknown type name 'suseconds_t'
   82 |     suseconds_t tv_usec;    /* and microseconds */
      |     ^~~~~~~~~~~
make: *** [rt-thread/components/libc/compilers/common/subdir.mk:33: rt-thread/components/libc/compilers/common/cunistd.o] Error 1
make: *** Waiting for unfinished jobs....
"make -j4 all2" terminated with exit code 2. Build might be incomplete.

12:48:07 Build Failed. 3 errors, 0 warnings. (took 2s.814ms)

seanfight, 2023-11-06:
> 12:48:04 **** Build of configuration Debug for project projrct **** make -j4 all2 riscv64-unknown-elf-gcc "../rt-thread/src/clock.c" riscv64-unknown-elf-gcc "../rt-thread/src/components.c" [...]

you can change the version of rtthread

Akashsujanapal, 2023-11-07:
bro thats problem was solved...but how to uplode the code in the chip?..explain pls

zhuzhengjing, 2024-05-08:
> ```shell
> CC build/kernel/src/clock.o
> CC build/kernel/src/components.o
> CC build/kernel/src/cpu.o
> CC build/kernel/src/device.o
> CC build/kernel/src/idle.o
> CC build/kernel/src/ipc.o
> CC build/kernel/src/irq.o
> CC build/kernel/src/kservice.o
> CC build/kernel/src/mem.o
> CC build/kernel/src/memheap.o
> CC build/kernel/src/mempool.o
> CC build/kernel/src/object.o
> CC build/kernel/src/scheduler.o
> CC build/kernel/src/thread.o
> CC build/kernel/src/timer.o
> LINK rtthread.elf
> aarch64-none-elf-objcopy -O binary rtthread.elf rtthread.bin
> aarch64-none-elf-size rtthread.elf
>    text	   data	    bss	    dec	    hex	filename
>  297258	   3600	  90608	 391466	  5f92a	rtthread.elf
> scons: done building targets.
> ~/workspace/rt-thread/bsp/qemu-virt64-aarch64$ aarch64-none-elf-gcc -v
> Using built-in specs.
> COLLECT_GCC=aarch64-none-elf-gcc
> COLLECT_LTO_WRAPPER=/opt/gcc-arm-10.2-2020.11-x86_64-aarch64-none-elf/bin/../libexec/gcc/aarch64-none-elf/10.2.1/lto-wrapper
> Target: aarch64-none-elf
> Configured with: /tmp/dgboter/bbs/build02--cen7x86_64/buildbot/cen7x86_64--aarch64-none-elf/build/src/gcc/configure --target=aarch64-none-elf --prefix=/tmp/dgboter/bbs/build02--cen7x86_64/buildbot/cen7x86_64--aarch64-none-elf/build/build-aarch64-none-elf/install// --with-gmp=/tmp/dgboter/bbs/build02--cen7x86_64/buildbot/cen7x86_64--aarch64-none-elf/build/build-aarch64-none-elf/host-tools --with-mpfr=/tmp/dgboter/bbs/build02--cen7x86_64/buildbot/cen7x86_64--aarch64-none-elf/build/build-aarch64-none-elf/host-tools --with-mpc=/tmp/dgboter/bbs/build02--cen7x86_64/buildbot/cen7x86_64--aarch64-none-elf/build/build-aarch64-none-elf/host-tools --with-isl=/tmp/dgboter/bbs/build02--cen7x86_64/buildbot/cen7x86_64--aarch64-none-elf/build/build-aarch64-none-elf/host-tools --disable-shared --disable-nls --disable-threads --disable-tls --enable-checking=release --enable-languages=c,c++,fortran --with-newlib --with-pkgversion='GNU Toolchain for the A-profile Architecture 10.2-2020.11 (arm-10.16)' --with-bugurl=https://bugs.linaro.org/
> Thread model: single
> Supported LTO compression algorithms: zlib
> gcc version 10.2.1 20201103 (GNU Toolchain for the A-profile Architecture 10.2-2020.11 (arm-10.16)) 
> ```
> 
> 没问题呀，
> 
> ```shell
> :~/workspace/rt-thread/bsp/qemu-virt64-aarch64$ git status . 
> 位于分支 master
> 您的分支与上游分支 'origin/master' 一致。
> 
> 无文件要提交，干净的工作区
> ```

最新的5.1.0也出现同样的问题，请问这个问题是什么原因导致的？

zhuzhengjing, 2024-05-08:
参考这里：https://club.rt-thread.org/ask/question/981978b1d833e9cf.html
需要设置 `_POSIX_C_SOURCE=1`