RT-Thread/rt-thread #6903: [ch32v307] 软件模拟的i2c与os tick产生耦合，读写出错

kylongmu, 2023-02-05:
git上拉取的最新rtthread，ch32v307
想着用不到每秒1000次的系统tick，改为100次以后，sht41，bmp180读取都不正确了，改回来就ok。难道是底层软件方式实现的i2c对os tick产生了耦合？为啥不直接做成硬件的i2c库呢？
感觉问题是i2c的频率400k，在一次读写时间超过10ms时，那么这样就会遇到os tick切换线程，线程被切换后软件模拟的i2c就会最长被挂起超10ms，导致未完成一次操作的i2c子设备判断超时退出i2c总线时序。
这也说明了软件模拟的i2c存在此类隐患，即使是1000次的tick，但是遇到优先级高连续多次被抢占的线程出现，那么i2c时序的完整性还是不能保证。

i-jaffer, 2023-02-12:
#6644

kylongmu, 2023-02-12:
> 此bug已修复，具体可查看#6644

这个问题是我最新拉取的rt-thread出现的，你的修复是Dec 6, 2022，是又有了更新了吗？

i-jaffer, 2023-02-12:
1. 请再次拉取，确认拉取到了最新的代码
2. 确认分支是在master分支，此修复目前只在master分支生效
3. 请标注具体代码位置，可在github上打开相应代码位置，点击代码行左侧，可复制对应代码行链接

kylongmu, 2023-02-13:
> 1. 请再次拉取，确认拉取到了最新的代码
> 2. 确认分支是在master分支，此修复目前只在master分支生效
> 3. 请标注具体代码位置，可在github上打开相应代码位置，点击代码行左侧，可复制对应代码行链接

[image]
检查了，是最新的，代码部分也一致。此问题看来超过了这次修改的范围。

mysterywolf, 2023-02-13:
问题在这里 https://github.com/RT-Thread/rt-thread/blob/master/bsp/wch/risc-v/Libraries/ch32_drivers/drv_soft_i2c.c#L118 wch的软件i2c延时使用的是systick，默认按照systick=1000处理的，此处应该直接调用rt_hw_delay_us会更好。

i-jaffer, 2023-02-13:
@kylongmu 可以参考这个，在对应bsp修改为硬件延时 https://github.com/RT-Thread/rt-thread/blob/fc1aced66574e6ef144b6b228978e734456e6f95/bsp/gd32303e-eval/drivers/drv_i2c.c#L92

closed 2024-03-08 by mysterywolf