RT-Thread/rt-thread #6922: master分支代码在Linux下clone下来，默认被修改

BernardXiong, 2023-02-09:
```bash
git status 
位于分支 master
您的分支与上游分支 'origin/master' 一致。

尚未暂存以备提交的变更：
  （使用 "git add <文件>..." 更新要提交的内容）
  （使用 "git restore <文件>..." 丢弃工作区的改动）
	修改：     bsp/imxrt/libraries/MIMXRT1170/MIMXRT1176/drivers/fsl_dac12.h
	修改：     bsp/imxrt/libraries/MIMXRT1170/MIMXRT1176/drivers/fsl_lpadc.h
	修改：     bsp/imxrt/libraries/MIMXRT1170/MIMXRT1176/drivers/fsl_semc.h
	修改：     bsp/imxrt/libraries/MIMXRT1170/MIMXRT1176/utilities/debug_console/fsl_debug_console_conf.h
	修改：     bsp/n32g452xx/Libraries/rt_drivers/drv_crypto.c
	修改：     bsp/n32g452xx/Libraries/rt_drivers/drv_crypto.h
```

估计是分行符的问题。

mysterywolf, 2023-02-09:
晚上我扫一遍

mysterywolf, 2023-02-11:
https://github.com/RT-Thread/rt-thread/pull/6925

closed 2023-02-15 by mysterywolf