RT-Thread/rt-thread #6929: 引用头文件 <sys/errno.h> 不符合 POSIX 标准导致工具链警告

polarvid, 2023-02-14:
仓库中部分文件存在类似的代码导致 rt-smart musl 工具链给出警告 `redirecting incorrect #include <sys/errno.h> to <errno.h>`

```c
#include <sys/errno.h>
```

根据 POSIX 标准<https://pubs.opengroup.org/onlinepubs/9699919799/basedefs/errno.h.html>应当使用 `#include <errno.h>`。这里这样使用如果是为了兼容旧工具链是否有更好的处理办法呢？

mysterywolf, 2023-02-14:
让工具链那边吧这个warning注释掉

这个没有办法，因为还要考虑到其他编译链，比如Keil IAR以及win32

polarvid, 2023-02-14:
增加一个兼容用的头文件，类似 `adapt_errno.h` 或专门放在一个兼容用的目录下 `adapt/errno.h`，在里面处理这些问题是否可行呢？

mysterywolf, 2023-02-14:
基本不可行，这些方案在一年半之前都论证过。改libc是一个眼高手低的活，需要亲自验证要保证所有工具链极其自带libc都可以编译过

mysterywolf, 2023-02-14:
https://www.rt-thread.org/document/site/#/rt-thread-version/rt-thread-standard/programming-manual/libc/iso-ansi-c?id=_71-%e4%b8%8d%e5%90%8c%e7%bc%96%e8%af%91%e5%b9%b3%e5%8f%b0%e7%9a%84%e5%85%bc%e5%ae%b9

mysterywolf, 2023-02-14:
有时间可以开会讨论一下 看看实际musl这边遇到什么问题了

BernardXiong, 2023-02-15:
这块可以对 musl 做个小调整