RT-Thread/rt-thread #6938: Context Switching is not happening

Embetronicx, 2023-02-18:
I have created two threads with the same priority. And the time slice is 1000 ticks. So, the context switch should happen after 2000 ticks. But only one task is running. Check the below example code.

```
#include <rtthread.h>

#define DBG_TAG "main"
#define DBG_LVL DBG_LOG

#include <rtdbg.h>
#include <drv_common.h>
#include <rtdevice.h>

static rt_thread_t threadId1 = RT_NULL;
static rt_thread_t threadId2 = RT_NULL;

/* Thread 1 */
static void thread1_function(void *parameter)
{
    while (1)
    {
        LOG_D("Thread 1\n");
    }
}

/* Thread 2 */
static void thread2_function(void *parameter)
{
    while (1)
    {
        LOG_D("Thread 2\n");
    }
}

int main(void)
{
    /* Create thread 1 */
    threadId1 = rt_thread_create("thread1_fn",  //Name
                            thread1_function,   //Function address
                            RT_NULL,            //Thread function parameter
                            1024,               //Stack Size
                            1,                  //Thread priority
                            1000);                 //Time slice in ticks

    /* Create thread 2 */
    threadId2 = rt_thread_create("thread2_fn",  //Name
                            thread2_function,   //Function address
                            RT_NULL,            //Thread function parameter
                            1024,               //Stack Size
                            1,                  //Thread priority
                            1000);                 //Time slice in ticks

    /* Start both threads */
    rt_thread_startup(threadId1);
    rt_thread_startup(threadId2);

    while(1);

    return RT_EOK;
}
```

mysterywolf, 2023-02-18:
you miss the rt_thread_mdelay function. You need to invoke the block functions.

Embetronicx, 2023-02-19:
If I use that mdelay function, it will put that thred into the sleep state
for that time. So the next thred will get cpu to run. But then what is the
use of that time Slice parameter? The time slice should work even if i
don't put mdelay. Both the three has to get the cpu based on round robin
method as both are having same priority.

Embetronicx, 2023-02-19:
I think i should call this rt_thread_startup(threadId2) from the thread 1.
Am I right?

mysterywolf, 2023-02-19:
Yes you are right, thread2 should be started up from thread1

closed 2023-02-20 by Embetronicx