RT-Thread/rt-thread #6953: RT-Thread Studio编译psoc6-cy8cproto-062S3-4343W bsp报错error: unknown type name 'clock_t'

dalen0822, 2023-02-22:
使用最新的rt-thread master分支代码，根据官方的指导文档操作编译Infineon PSoc6系统的BSP代码编译不通过。
操作如下：
在bsp/Infineon/psoc6-cy8cproto-062S3-4343W 目录下执行 scons --dist 生成目标代码，使用RT-Thread Studio 导入后，执行编译会出现如下错误，请问如何解决。
`17:05:27 **** Build of configuration Debug for project project ****
make -j8 all 
arm-none-eabi-gcc "../rt-thread/src/clock.c"
arm-none-eabi-gcc "../rt-thread/src/components.c"
arm-none-eabi-gcc "../rt-thread/src/device.c"
arm-none-eabi-gcc "../rt-thread/src/idle.c"
arm-none-eabi-gcc "../rt-thread/src/ipc.c"
arm-none-eabi-gcc "../rt-thread/src/irq.c"
arm-none-eabi-gcc "../rt-thread/src/kservice.c"
arm-none-eabi-gcc "../rt-thread/src/mem.c"
arm-none-eabi-gcc "../rt-thread/src/mempool.c"
arm-none-eabi-gcc "../rt-thread/src/object.c"
arm-none-eabi-gcc "../rt-thread/src/scheduler.c"
arm-none-eabi-gcc "../rt-thread/src/thread.c"
arm-none-eabi-gcc "../rt-thread/src/timer.c"
arm-none-eabi-gcc "../rt-thread/libcpu/arm/cortex-m4/context_gcc.S"
arm-none-eabi-gcc "../rt-thread/libcpu/arm/cortex-m4/cpuport.c"
arm-none-eabi-gcc "../rt-thread/libcpu/arm/common/div0.c"
arm-none-eabi-gcc "../rt-thread/libcpu/arm/common/showmem.c"
arm-none-eabi-gcc "../rt-thread/components/utilities/libadt/avl.c"
arm-none-eabi-gcc "../rt-thread/components/libc/compilers/newlib/syscalls.c"
arm-none-eabi-gcc "../rt-thread/components/libc/compilers/common/cctype.c"
arm-none-eabi-gcc "../rt-thread/components/libc/compilers/common/cstdio.c"
arm-none-eabi-gcc "../rt-thread/components/libc/compilers/common/cstdlib.c"
arm-none-eabi-gcc "../rt-thread/components/libc/compilers/common/cstring.c"
arm-none-eabi-gcc "../rt-thread/components/libc/compilers/common/ctime.c"
arm-none-eabi-gcc "../rt-thread/components/libc/compilers/common/cwchar.c"
arm-none-eabi-gcc "../rt-thread/components/finsh/cmd.c"
arm-none-eabi-gcc "../rt-thread/components/finsh/msh.c"
arm-none-eabi-gcc "../rt-thread/components/finsh/msh_parse.c"
arm-none-eabi-gcc "../rt-thread/components/finsh/shell.c"
arm-none-eabi-gcc "../rt-thread/components/drivers/serial/serial.c"
arm-none-eabi-gcc "../rt-thread/components/drivers/misc/pin.c"
arm-none-eabi-gcc "../rt-thread/components/drivers/misc/rt_drv_pwm.c"
arm-none-eabi-gcc "../rt-thread/components/drivers/ipc/completion.c"
In file included from Z:\rt-thread\rt-thread\bsp\Infineon\psoc6-cy8cproto-062S3-4343W\dist\project\rt-thread\components\libc\compilers\common\include/sys/time.h:18:0,
                 from Z:\rt-thread\rt-thread\bsp\Infineon\psoc6-cy8cproto-062S3-4343W\dist\project\rt-thread\components\libc\compilers\common\include/sys/select.h:17,
                 from c:\rt-threadstudio\repo\extract\toolchain_support_packages\arm\gnu_tools_for_arm_embedded_processors\5.4.1\arm-none-eabi\include\sys\types.h:68,
                 from c:\rt-threadstudio\repo\extract\toolchain_support_packages\arm\gnu_tools_for_arm_embedded_processors\5.4.1\arm-none-eabi\include\stdio.h:61,
                 from ../rt-thread/components/libc/compilers/newlib/syscalls.c:16:
c:\rt-threadstudio\repo\extract\toolchain_support_packages\arm\gnu_tools_for_arm_embedded_processors\5.4.1\arm-none-eabi\include\time.h:56:1: error: unknown type name 'clock_t'
 clock_t    _EXFUN(clock,    (void));
 ^
In file included from Z:\rt-thread\rt-thread\bsp\Infineon\psoc6-cy8cproto-062S3-4343W\dist\project\rt-thread\components\libc\compilers\common\include/sys/time.h:18:0,
                 from Z:\rt-thread\rt-thread\bsp\Infineon\psoc6-cy8cproto-062S3-4343W\dist\project\rt-thread\components\libc\compilers\common\include/sys/select.h:17,
                 from c:\rt-threadstudio\repo\extract\toolchain_support_packages\arm\gnu_tools_for_arm_embedded_processors\5.4.1\arm-none-eabi\include\sys\types.h:68,
                 from c:\rt-threadstudio\repo\extract\toolchain_support_packages\arm\gnu_tools_for_arm_embedded_processors\5.4.1\arm-none-eabi\include\stdio.h:61,
                 from ../rt-thread/components/finsh/shell.c:23:
c:\rt-threadstudio\repo\extract\toolchain_support_packages\arm\gnu_tools_for_arm_embedded_processors\5.4.1\arm-none-eabi\include\time.h:56:1: error: unknown type name 'clock_t'
 clock_t    _EXFUN(clock,    (void));
 ^
In file included from Z:\rt-thread\rt-thread\bsp\Infineon\psoc6-cy8cproto-062S3-4343W\dist\project\rt-thread\components\libc\compilers\common\include/sys/select.h:17:0,
                 from c:\rt-threadstudio\repo\extract\toolchain_support_packages\arm\gnu_tools_for_arm_embedded_processors\5.4.1\arm-none-eabi\include\sys\types.h:68,
                 from c:\rt-threadstudio\repo\extract\toolchain_support_packages\arm\gnu_tools_for_arm_embedded_processors\5.4.1\arm-none-eabi\include\stdio.h:61,
                 from ../rt-thread/components/finsh/shell.c:23:
Z:\rt-thread\rt-thread\bsp\Infineon\psoc6-cy8cproto-062S3-4343W\dist\project\rt-thread\components\libc\compilers\common\include/sys/time.h:56:5: error: unknown type name 'suseconds_t'
     suseconds_t tv_usec;    /* and microseconds */
     ^
In file included from Z:\rt-thread\rt-thread\bsp\Infineon\psoc6-cy8cproto-062S3-4343W\dist\project\rt-thread\components\libc\compilers\common\include/sys/select.h:17:0,
                 from c:\rt-threadstudio\repo\extract\toolchain_support_packages\arm\gnu_tools_for_arm_embedded_processors\5.4.1\arm-none-eabi\include\sys\types.h:68,
                 from c:\rt-threadstudio\repo\extract\toolchain_support_packages\arm\gnu_tools_for_arm_embedded_processors\5.4.1\arm-none-eabi\include\stdio.h:61,
                 from ../rt-thread/components/libc/compilers/newlib/syscalls.c:16:
Z:\rt-thread\rt-thread\bsp\Infineon\psoc6-cy8cproto-062S3-4343W\dist\project\rt-thread\components\libc\compilers\common\include/sys/time.h:56:5: error: unknown type name 'suseconds_t'
     suseconds_t tv_usec;    /* and microseconds */
     ^
In file included from Z:\rt-thread\rt-thread\bsp\Infineon\psoc6-cy8cproto-062S3-4343W\dist\project\rt-thread\components\libc\compilers\common\include/sys/time.h:18:0,
                 from Z:\rt-thread\rt-thread\bsp\Infineon\psoc6-cy8cproto-062S3-4343W\dist\project\rt-thread\components\libc\compilers\common\include/sys/select.h:17,
[... 4 lines cut ...]
c:\rt-threadstudio\repo\extract\toolchain_support_packages\arm\gnu_tools_for_arm_embedded_processors\5.4.1\arm-none-eabi\include\time.h:56:1: error: unknown type name 'clock_t'
 clock_t    _EXFUN(clock,    (void));
 ^
In file included from Z:\rt-thread\rt-thread\bsp\Infineon\psoc6-cy8cproto-062S3-4343W\dist\project\rt-thread\components\libc\compilers\common\include/sys/select.h:17:0,
                 from c:\rt-threadstudio\repo\extract\toolchain_support_packages\arm\gnu_tools_for_arm_embedded_processors\5.4.1\arm-none-eabi\include\sys\types.h:68,
                 from c:\rt-threadstudio\repo\extract\toolchain_support_packages\arm\gnu_tools_for_arm_embedded_processors\5.4.1\arm-none-eabi\include\stdio.h:61,
                 from Z:\rt-thread\rt-thread\bsp\Infineon\psoc6-cy8cproto-062S3-4343W\dist\project\rt-thread\components\libc\compilers\common\include/posix/stdio.h:18,
                 from ../rt-thread/components/libc/compilers/common/cstdio.c:11:
Z:\rt-thread\rt-thread\bsp\Infineon\psoc6-cy8cproto-062S3-4343W\dist\project\rt-thread\components\libc\compilers\common\include/sys/time.h:56:5: error: unknown type name 'suseconds_t'
     suseconds_t tv_usec;    /* and microseconds */
     ^
make: *** [rt-thread/components/finsh/subdir.mk:27: rt-thread/components/finsh/shell.o] Error 1
make: *** Waiting for unfinished jobs....
make: *** [rt-thread/components/libc/compilers/common/subdir.mk:33: rt-thread/components/libc/compilers/common/cstdio.o] Error 1
make: *** [rt-thread/components/libc/compilers/newlib/subdir.mk:18: rt-thread/components/libc/compilers/newlib/syscalls.o] Error 1
../rt-thread/components/drivers/misc/rt_drv_pwm.c: In function '_pwm_control':
../rt-thread/components/drivers/misc/rt_drv_pwm.c:25:14: error: 'PWMN_CMD_ENABLE' undeclared (first use in this function)
         case PWMN_CMD_ENABLE:
              ^
../rt-thread/components/drivers/misc/rt_drv_pwm.c:25:14: note: each undeclared identifier is reported only once for each function it appears in
../rt-thread/components/drivers/misc/rt_drv_pwm.c:26:26: error: dereferencing pointer to incomplete type 'struct rt_pwm_configuration'
             configuration->complementary = RT_TRUE;
                          ^
../rt-thread/components/drivers/misc/rt_drv_pwm.c:28:14: error: 'PWMN_CMD_DISABLE' undeclared (first use in this function)
         case PWMN_CMD_DISABLE:`

zhkag, 2023-02-22:
[image]

试试用这个功能导入到 stdio

dalen0822, 2023-02-22:
> [image]
> 
> 试试用这个功能导入到 stdio

仍然有同样的问题。
在linux下使用scons 编译就不会出现上述的报错。

zhkag, 2023-02-22:
> > [image]
> > 试试用这个功能导入到 stdio
> 
> 仍然有同样的问题。 在linux下使用scons 编译就不会出现上述的报错。

您需要 dist 生成的目录下重新使用 scons --target=eclipse 生成工程后然后再导入到 stdio 中

dalen0822, 2023-02-22:
> > > [image]
> > > 试试用这个功能导入到 stdio
> > 
> > 
> > 仍然有同样的问题。 在linux下使用scons 编译就不会出现上述的报错。
> 
> 您需要 dist 生成的目录下重新使用 scons --target=eclipse 生成工程后然后再导入到 stdio 中

非常感谢，问题解决了。

closed 2023-02-22 by dalen0822

mysterywolf, 2023-03-20:
https://github.com/RT-Thread/rt-thread/pull/7073