RT-Thread/rt-thread #6965: RT-Thread Cache 处理优化

Guozhanxin, 2023-02-23:
驱动目前对于 Cache 处理是耦合在驱动代码里的。是不是将来抽象出来一层，独立管理 DMA 与驱动 有关的 Buffer
[image]
### 方案一
驱动对应用传进来的buf，做一次拷贝，拷贝到满足要求的内存空间再进行DMA操作。
[image]

### 方案二
扩充 control 接口，应用直接向驱动申请满足要求的驱动
[image]

[image]

BernardXiong, 2023-02-27:
rthw.h定义了几个cache ops:
```c
void rt_hw_cpu_icache_enable(void);
void rt_hw_cpu_icache_disable(void);
rt_base_t rt_hw_cpu_icache_status(void);
void rt_hw_cpu_icache_ops(int ops, void* addr, int size);

void rt_hw_cpu_dcache_enable(void);
void rt_hw_cpu_dcache_disable(void);
rt_base_t rt_hw_cpu_dcache_status(void);
void rt_hw_cpu_dcache_ops(int ops, void* addr, int size);
```

wosayttn, 2023-03-03:
某平台的 drv_uart.c for serial_v1 驅動實現內，搞了個偷天換日，Zero-copy 和排除了 invalidate un-cleaned data 問題。

drv_uart.c
```
#if defined(BSP_USING_MMU)
        /* Allocate aligned and align-up DMA buffer for cache coherence. */
        psNuUart->dmabuf.pu8RxBuf = rt_malloc_align(RT_ALIGN(i32TriggerLen, 32), 32);
        psNuUart->dmabuf.bufsize = 0;
        if (psNuUart->dmabuf.pu8RxBuf == RT_NULL)
        {
            LOG_E("Failed to allocate dma memory %d.", i32TriggerLen);
            goto exit_nu_pdma_uart_rx_config;
        }
        rx_fifo->buffer = (rt_uint8_t*) psNuUart->dmabuf.pu8RxBuf;
        rt_memset(rx_fifo->buffer, 0, i32TriggerLen);
#endif
        psNuUart->dmabuf.bufsize = i32TriggerLen;
        psNuUart->dmabuf.put_index = 0;
```

serial 內，申請的 RX Buffer時，若符合 **rt_malloc_align(RT_ALIGN( BUFSIZE, CACHE_LINE_SIZE), CACHE_LINE_SIZE)** ， 驅動層就不需要偷天換日了。