RT-Thread/rt-thread #6971: rt_sem_take为何能在SMP环境下使用？

FragrantRye, 2023-02-24:
阅读源码的过程中对这段代码有疑问，烦请各位老师解答！

在_rt_sem_take函数中，对value的读写如何保证原子性的？
SMP环境下，如果value变量的read-modify-write操作被其他处理器打断了，应该会出现两个处理器同时拿到信号量的问题吧？
```C
static rt_err_t _rt_sem_take(rt_sem_t sem, rt_int32_t timeout, int suspend_flag)
{
    // 中间省略
    level = rt_hw_interrupt_disable();
    if (sem->value > 0)
    {
        /* semaphore is available */
        sem->value --;

        /* enable interrupt */
        rt_hw_interrupt_enable(level);
    }
```

FragrantRye, 2023-02-24:
另外，在linux源码中(kernel/locking/semaphore.c)，这部分代码是使用了spinlock来保护的，供参考
```C
/**
 * down_interruptible - acquire the semaphore unless interrupted
 * @sem: the semaphore to be acquired
 *
 * Attempts to acquire the semaphore.  If no more tasks are allowed to
 * acquire the semaphore, calling this function will put the task to sleep.
 * If the sleep is interrupted by a signal, this function will return -EINTR.
 * If the semaphore is successfully acquired, this function returns 0.
 */
int down_interruptible(struct semaphore *sem)
{
	unsigned long flags;
	int result = 0;

	might_sleep();
	raw_spin_lock_irqsave(&sem->lock, flags);
	if (likely(sem->count > 0))
		sem->count--;
	else
		result = __down_interruptible(sem);
	raw_spin_unlock_irqrestore(&sem->lock, flags);

	return result;
}
EXPORT_SYMBOL(down_interruptible);
```

BernardXiong, 2023-02-27:
可以看下rt_hw_interrupt_disable在多核下的实现。

FragrantRye, 2023-02-27:
懂了，rt_hw_interrupt_disable在多核下实际调用了rt_hw_spin_lock()
感谢回复！

closed 2023-02-27 by FragrantRye