RT-Thread/rt-thread #6985: 合并lwIP版本后ipv6的问题

BernardXiong, 2023-02-27:
合并lwIP版本后，在目前代码中是共用一份lwipopts.h文件，但在这个配置中，当打开IPv6后编译失败。

```bash
CC build/kernel/components/net/lwip/lwip-2.0.3/src/core/init.o
/home/workspace/rt-thread/components/net/lwip/lwip-2.0.3/src/core/init.c:134:4: error: #error "MEMP_NUM_SYS_TIMEOUT is too low to accomodate all required timeouts"
  134 |   #error "MEMP_NUM_SYS_TIMEOUT is too low to accomodate all required timeouts"
      |    ^~~~~

```

MEMP_NUM_SYS_TIMEOUT定义过小。

BernardXiong, 2023-02-27:
lwIP 2.0.3/2.1.2都存在这样的问题

flyingcys, 2023-04-13:
> 合并lwIP版本后，在目前代码中是共用一份lwipopts.h文件，但在这个配置中，当打开IPv6后编译失败。
> 
> ```shell
> CC build/kernel/components/net/lwip/lwip-2.0.3/src/core/init.o
> /home/workspace/rt-thread/components/net/lwip/lwip-2.0.3/src/core/init.c:134:4: error: #error "MEMP_NUM_SYS_TIMEOUT is too low to accomodate all required timeouts"
>   134 |   #error "MEMP_NUM_SYS_TIMEOUT is too low to accomodate all required timeouts"
>       |    ^~~~~
> ```
> 
> MEMP_NUM_SYS_TIMEOUT定义过小。

MEMP_NUM_SYS_TIMEOUT比较容易修复，目前还发现一个问题：在ip6_addr.h中LWIP_IPV6_ADDRESS_LIFETIMES宏没有被定义，
```shell
#if LWIP_IPV6_ADDRESS_LIFETIMES
#define IP6_ADDR_LIFE_STATIC   (0)
#define IP6_ADDR_LIFE_INFINITE (0xffffffffUL)
#define ip6_addr_life_isstatic(addr_life) ((addr_life) == IP6_ADDR_LIFE_STATIC)
#define ip6_addr_life_isinfinite(addr_life) ((addr_life) == IP6_ADDR_LIFE_INFINITE)
#endif /* LWIP_IPV6_ADDRESS_LIFETIMES */
```
但是实际上已经在opt.h中定义了，看起来好像是头文件被重复调用起来的，暂时没有好的方法定位

xiaoniaoyou6hua, 2023-04-19:
目前就本人使用2.1.2版本中，开启ipv6后编译第一个error是MEMP_NUM_SYS_TIMEOUT，这里据情况适当配置即可解决；第二个error就是IP6_ADDR_LIFE_STATIC变量undefined；第三个是ethernetif.c中local.sin_addr.s_addr = (netif->ip_addr.addr);该句报has no member named 'addr'这是源于开启ipv6后，伴随着开启了一些宏，宏作用之后相关结构体成员改变。解决完编译问题上板子发现跑不通，定位到问题在于不同.c下netdev结构体存在差异，导致异常，具体的差异存在于结构体成员中ip_addr_t类型的成员。这也是楼上所提及的问题。
    在ip6_addr.h中LWIP_IPV6_ADDRESS_LIFETIMES宏没有被定义的问题是这样的，在opt.h中包含了lwipops.h，lwipops.h中末尾又包含了ip_addr.h，ip_addr.h又包含了ip6_addr.h。而ip6_addr中依赖来自opt.h中的此时宏尚未定义，即ip6_addr.h被率先隐式包含进来，此时宏尚未配置。上述中IP6_ADDR_LIFE_STATIC变量undefined的问题也是源于此，ip6_addr.h被率先隐私包含相关宏没有拿到没有去定义相关变量。
    has no member named 'addr'这个error在开启ipv6后会出现，这个问题在2.0.3版本中也会出现。这里应该增加宏控制，分情况去拿结构体中的相关成员，来满足ipv6开启与否不同情况下的功能需求。

xiaoniaoyou6hua, 2023-04-19:
### 目前就本人使用2.1.2版本中，开启ipv6后编译第一个error是MEMP_NUM_SYS_TIMEOUT，这里据情况适当配置即可解决；第二个error就是IP6_ADDR_LIFE_STATIC变量undefined；第三个是ethernetif.c中local.sin_addr.s_addr = (netif->ip_addr.addr);该句报has no member named 'addr'这是源于开启ipv6后，伴随着开启了一些宏，宏作用之后相关结构体成员改变。解决完编译问题上板子发现跑不通，定位到问题在于不同.c下netdev结构体存在差异，导致异常，具体的差异存在于结构体成员中ip_addr_t类型的成员。这也是楼上所提及的问题。
在ip6_addr.h中LWIP_IPV6_ADDRESS_LIFETIMES宏没有被定义的问题是这样的，在opt.h中包含了lwipops.h，lwipops.h中末尾又包含了ip_addr.h，ip_addr.h又包含了ip6_addr.h。而ip6_addr中依赖来自opt.h中的此时宏尚未定义，即ip6_addr.h被率先隐式包含进来，此时宏尚未配置。上述中IP6_ADDR_LIFE_STATIC变量undefined的问题也是源于此，ip6_addr.h被率先隐私包含相关宏没有拿到没有去定义相关变量。
has no member named 'addr'这个error在开启ipv6后会出现，这个问题在2.0.3版本中也会出现。这里应该增加宏控制，分情况去拿结构体中的相关成员，来满足ipv6开启与否不同情况下的功能需求。

xiaoniaoyou6hua, 2023-04-19:
​		目前就本人使用`2.1.2`版本中，开启```ipv6```后编译。

​		第一个```error```是`MEMP_NUM_SYS_TIMEOUT`，这里据情况适当配置即可解决；

​		第二个```error```就是`IP6_ADDR_LIFE_STATIC`变量```undefined```；

​		第三个```error```是`ethernetif.c`中`local.sin_addr.s_addr = (netif->ip_addr.addr)`。该句报`has no member named 'addr'`这是源于开启```ipv6```后，伴随着开启了一些宏，宏作用之后相关结构体成员改变。这个```error```在开启```ipv6```后会出现，这个问题在`2.0.3`版本中也会出现。这里应该增加宏控制，分情况去拿结构体中的相关成员，来满足```ipv6```开启与否不同情况下的功能需求。

​		解决完编译问题上板子发现跑不通，定位到问题在于不同```.c```下`netdev`结构体存在差异，导致异常，具体的差异存在于`netdev`结构体成员中`ip_addr_t`类型的成员。这也是楼上所提及的跟宏缺失相关的问题。

​		在`ip6_addr.h`中`LWIP_IPV6_ADDRESS_LIFETIMES`宏没有被定义的问题是这样的，在`opt.h`中包含了`lwipops.h`，`lwipops.h`中末尾又包含了`ip_addr.h`，`ip_addr.h`又包含了`ip6_addr.h`。而`ip6_add`r中依赖来自`opt.h`中的宏此时尚未定义，即`ip6_addr.h`被率先隐式包含进来，此时宏尚未配置。

```
//lwipops.h末尾包含ip_addr.h

#include "lwip/ip_addr.h"
struct netif *lwip_ip4_route_src(const ip4_addr_t *dest, const ip4_addr_t *src);
```

​		上述中编译的第二个```erroe``` `IP6_ADDR_LIFE_STATIC`变量```undefined```的问题也是源于此，`ip6_addr.h`被率先隐式包含后，相关宏还没有拿到，致使相关变量未定义。

xiaoniaoyou6hua, 2023-04-19:
​		目前就本人使用`2.1.2`版本中，开启```ipv6```后编译。

​		第一个```error```是`MEMP_NUM_SYS_TIMEOUT`，这里据情况适当配置即可解决；

​		第二个```error```就是`IP6_ADDR_LIFE_STATIC`变量```undefined```；

​		第三个```error```是`ethernetif.c`中`local.sin_addr.s_addr = (netif->ip_addr.addr)`。该句报`has no member named 'addr'`这是源于开启```ipv6```后，伴随着开启了一些宏，宏作用之后相关结构体成员改变。这个```error```在开启```ipv6```后会出现，这个问题在`2.0.3`版本中也会出现。这里应该增加宏控制，分情况去拿结构体中的相关成员，来满足```ipv6```开启与否不同情况下的功能需求。

​		解决完编译问题上板子发现跑不通，定位到问题在于不同```.c```下`netdev`结构体存在差异，导致异常，具体的差异存在于`netdev`结构体成员中`ip_addr_t`类型的成员。这也是楼上所提及的跟宏缺失相关的问题。

​		在`ip6_addr.h`中`LWIP_IPV6_ADDRESS_LIFETIMES`宏没有被定义的问题是这样的，在`opt.h`中包含了`lwipops.h`，`lwipops.h`中末尾又包含了`ip_addr.h`，`ip_addr.h`又包含了`ip6_addr.h`。而`ip6_add`r中依赖来自`opt.h`中的宏此时尚未定义，即`ip6_addr.h`被率先隐式包含进来，此时宏尚未配置。

```
//lwipops.h末尾包含ip_addr.h

#include "lwip/ip_addr.h"
struct netif *lwip_ip4_route_src(const ip4_addr_t *dest, const ip4_addr_t *src);
```

​		上述中编译的第二个```error``` `IP6_ADDR_LIFE_STATIC`变量```undefined```的问题也是源于此，`ip6_addr.h`被率先隐式包含后，相关宏还没有拿到，致使相关变量未定义。

closed 2023-06-12 by Guozhanxin